thaytan/gst-rpicamsrc #16: Some missing properties

towolf, 2015-02-11:
Works fine, thank you.

When I last used the camera I used v4l2src with the "collabora" v4l2 kernel driver. Now that stopped working (I get oops in dmesg and it got annoying) so I tried your approach and it does the job quite fine.

 1.  The auto white balance of the camera is very bad. It wobbles around all the time like being high on LSD. 

     They realized this and offer to fixate AWB gains now with `--awbg <red>,<blue>`

 2.  Another thing, [this guy implemented "Periodic Intra Refresh"](https://github.com/raspberrypi/userland/pull/215). I’m hoping this might get latency just a little lower.

 3.  And finally they have an explicit `--mode` switch for sensor operation modes. They switch modes implicitly based on width and height thresholds. An explicit selection would be good

Could these three things be added?

And if you don’t mind, some questions. Are the timestamps the same that the V4l2 driver delivers? Any hint what would be the best way to stream low latency video onto Android? Via WebRTC maybe?

thaytan, 2015-02-13:
Thanks! Good suggestions. I've been meaning to do some merging of the upstream RPi raspivid/raspistill changes, but not gotten to it. I'll try and do that soon.

Unfortunately atm, timestamping is based purely on when encoded frames emerge from the H.264 encoder. It would be good to timestamp them earlier in the chain, closer to the real capture timestamp.

RTSP with suitable low-latency jitterbuffer configured on the receiver or WebRTC would be my suggestions for streaming - but the latter might be trickier to get set up, although OpenWebRTC is coming along.

towolf, 2015-02-13:
Cool, I looked at most recent raspivid help and these things would be nice to have:

> raspivid
>
> raspivid Camera App v1.3.12
>
> Display camera output to display, and optionally saves an H264 capture at requested bitrate
>
>
> usage: raspivid [options]
>
> Image parameter commands
>
> -?, --help	: This help information
> -w, --width	: Set image width <size>. Default 1920
> -h, --height	: Set image height <size>. Default 1080
> -b, --bitrate	: Set bitrate. Use bits per second (e.g. 10MBits/s would be -b 10000000)
> -o, --output	: Output filename <filename> (to write to stdout, use '-o -')
> -v, --verbose	: Output verbose information during run
> -t, --timeout	: Time (in ms) to capture for. If not specified, set to 5s. Zero to disable
> -d, --demo	: Run a demo mode (cycle through range of camera options, no capture)
> -fps, --framerate	: Specify the frames per second to record
> -e, --penc	: Display preview image *after* encoding (shows compression artifacts)
> -g, --intra	: Specify the intra refresh period (key frame rate/GoP size). Zero to produce an initial I-frame and then just P-frames.
> **-pf, --profile	: Specify H264 profile to use for encoding**
> -td, --timed	: Cycle between capture and pause. -cycle on,off where on is record time and off is pause time in ms
> -s, --signal	: Cycle between capture and pause on Signal
> -k, --keypress	: Cycle between capture and pause on ENTER
> -i, --initial	: Initial state. Use 'record' or 'pause'. Default 'record'
> **-qp, --qp	: Quantisation parameter. Use approximately 10-40. Default 0 (off)**
> **-ih, --inline	: Insert inline headers (SPS, PPS) to stream**
> -sg, --segment	: Segment output file in to multiple files at specified interval <ms>
> -wr, --wrap	: In segment mode, wrap any numbered filename back to 1 when reach number
> -sn, --start	: In segment mode, start with specified segment number
> -sp, --split	: In wait mode, create new output file for each start event
> -c, --circular	: Run encoded data through circular buffer until triggered then save
> -x, --vectors	: Output filename <filename> for inline motion vectors
> -cs, --camselect	: Select camera <number>. Default 0
> -set, --settings	: Retrieve camera settings and write to stdout
> **-md, --mode	: Force sensor mode. 0=auto. See docs for other modes available**
> **-if, --irefresh	: Set intra refresh type**
>
>
> H264 Profile options :
> baseline,main,high
>
>
> H264 Intra refresh options :
> cyclic,adaptive,both,cyclicrows
>
> Preview parameter commands
>
> -p, --preview	: Preview window settings <'x,y,w,h'>
> -f, --fullscreen	: Fullscreen preview mode
> -op, --opacity	: Preview window opacity (0-255)
> -n, --nopreview	: Do not display a preview window
>
> Image parameter commands
>
> -sh, --sharpness	: Set image sharpness (-100 to 100)
> -co, --contrast	: Set image contrast (-100 to 100)
> -br, --brightness	: Set image brightness (0 to 100)
> -sa, --saturation	: Set image saturation (-100 to 100)
> -ISO, --ISO	: Set capture ISO
> -vs, --vstab	: Turn on video stabilisation
> -ev, --ev	: Set EV compensation
> -ex, --exposure	: Set exposure mode (see Notes)
> -awb, --awb	: Set AWB mode (see Notes)
> -ifx, --imxfx	: Set image effect (see Notes)
> -cfx, --colfx	: Set colour effect (U:V)
> -mm, --metering	: Set metering mode (see Notes)
> -rot, --rotation	: Set image rotation (0-359)
> -hf, --hflip	: Set horizontal flip
> -vf, --vflip	: Set vertical flip
> -roi, --roi	: Set region of interest (x,y,w,d as normalised coordinates [0.0-1.0])
> **-ss, --shutter	: Set shutter speed in microseconds**
> **-awbg, --awbgains	: Set AWB gains - AWB mode must be off**
> -drc, --drc	: Set DRC Level
> -st, --stats	: Force recomputation of statistics on stills capture pass
> **-a, --annotate	: Enable/Set annotate flags or text**
>
>

towolf, 2015-02-13:
About timestamps, [this guy](http://www.raspberrypi.org/forums/viewtopic.php?p=684870#p684870) made a [testing fork](https://github.com/ethanol100/userland/commit/b152b7dd40d89ff8df18e20e349c056e450c538f) that writes MMAL_PARAM_TIMESTAMP_MODE_RAW_STC timestamps to file.

The "collabora" v4l2 driver has [timestamps described as such](http://www.raspberrypi.org/forums/viewtopic.php?p=463864#p463864). Dunno if the mmal timestamps refer to the same.

towolf, 2015-02-14:
[More timestamp info](http://www.raspberrypi.org/forums/viewtopic.php?p=694050#p694050)

thaytan, 2015-03-05:
Timestamping still needs 'fixing' - although the dumb timestamping it does atm generally seems OK, and I need to implement setting the H.264 profile from what is negotiated with downstream.

The rest of the things that have been added upstream are now merged. Please test - I haven't tested exhaustively yet.

towolf, 2015-03-05:
That is a very comprehensive list of additions. Thank you.

I will test them all as soon as I get the chance.

thaytan, 2015-03-06:
commit c3e93bb and commit 196969 add the ability to set the encoder profile through caps negotiation with downstream. e.g.:

gst-launch-1.0 -e rpicamsrc ! video/x-h264,width=640,height=480,profile=main ! h264parse ! queue ! filesink location=~/test-main.h264

closed 2015-03-09 by thaytan

samskiter, 2016-01-07:
@thaytan so should AWB gains and exposure compensation be working? I'm not seeing any change after setting awb-mode=off and adjusting awb-gain-red/blue.

Similarly exposure-compensation.

Other props like brightness and saturation are working fine

commit f708a8ec (2015-03-09): Use MMAL PTS and STC to calculate GStreamer timestamps

Don't apply timestamps based on output time from the encoder,
but use the MMAL STC and capture PTS to generate a GStreamer
timestamp that more accurately resembles the input (and would
preserve reordering should the encoder ever add B-frames).

Fixes #16