thaytan/gst-rpicamsrc #18: failed to load

gregd72002, 2015-02-15:
Hi,

I'm trying to run it in a uClibc based system. It compiles well but when trying to load it reports error:

# gst-inspect-1.0 ./libgstrpicamsrc.so

gst-inspect-1.0: symbol 'vcos_pthreads_logging_assert': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_log_register': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_abort': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_generic_mem_free': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_thread_join': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_global_unlock': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_generic_mem_alloc': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_thread_create': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_log_impl': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_pthreads_map_errno': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_thread_attr_init': can't resolve symbol

gst-inspect-1.0: symbol 'vcos_global_lock': can't resolve symbol

(gst-inspect-1.0:2689): GStreamer-WARNING **: Failed to load plugin './libgstrpicamsrc.so': unknown dl-error
Could not load plugin file: Opening module failed: unknown dl-error

thaytan, 2015-03-06:
Did you find a solution to this? It seems like your LD_LIBRARY_PATH might be wrong and not picking up your /opt/vc/lib (or equivalent)?

thaytan, 2015-03-24:
Please re-open if you have any more info.

closed 2015-03-24 by thaytan