thaytan/gst-rpicamsrc #19: setting properties at runtime

sphaero, 2015-02-17:
I've tested some code which works with v4l2src but it doesn't with gst-rpicamsrc. Can anybody confirm whether this is an issue or...

sphaero, 2015-02-17:
I've set GST_DEBUG=rpicamsrc:6 
```
0:00:14.352468232 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3468
0:00:14.385715042 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3447
0:00:14.418840856 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3598
0:00:14.452051667 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3435
bd44b257-6ab0-43a6-bbd6-5b14e71958cd bd44b2 {'brightness': {'value': 50}} () {}
set brightness to 50
0:00:14.478879707 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 8318
0:00:14.518428291 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3655
0:00:14.551689100 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3668
0:00:14.584937910 31686  0x23b85b0 LOG                rpicamsrc gstrpicamsrc.c:697:gst_rpi_cam_src_create:<videosrc0> Made buffer of size 3700
0:00:14.615352821 31686  0x23b85b0 LOG                rpicamsrc 
```

At start it does seem to get it:
```
0:00:00.830701262 31707   0xf7a5b0 LOG                rpicamsrc RaspiCamControl.c:584:raspicamcontrol_dump_parameters: Sharpness 0, Contrast 0, Brightness 10
```

thaytan, 2015-03-05:
There's not enough info here for me to make any guesses. What is your working vs not-working test case?
From those logs, it looks like rpicamsrc is producing buffers.

sphaero, 2015-03-05:
For example run a pipeline including the rpicamsrc and then try to modify the saturation parameter at runtime. I'll see if I can provide a simple python example.

sphaero, 2015-03-05:
K, here's example code illustrating the problem
https://gist.github.com/sphaero/d7747dce4cd872747910

This works for a v4l2src but it doesn't for rpicamsrc. Just uncomment line 43 for a rpicamsrc instead of a v4l2src

thaytan, 2015-03-05:
Ah, OK.

It should be possible to update some of the properties at runtime - ones that don't require reconfiguring the camera. I'll try it out.

thaytan, 2015-04-26:
Give that a try. I added a modified version of your test program under examples/dynamicprops.py - I hope that's OK.

commit 2f46fae6 (2015-04-26): Update properties dynamically where possible

Update camera and encoder properties at runtime
where possible

Fixes #19 and #23
closed 2015-04-26 by thaytan