thaytan/gst-rpicamsrc #20: Streaming over the network with TCP without gdppay

nh2, 2015-02-27:
Hey,

this is not really an issue with rpicamsrc, rather a question.

`rpicamsrc` is *by far* the most CPU efficient way to get video out of the Pi that I've found, and I'd like to make it available over the network.

I can do it via **UDP** as described [here](http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=44675#p636710):

**pi:** `gst-launch-1.0 rpicamsrc bitrate=1000000 ! 'video/x-h264, width=1280, height=720, framerate=30/1' ! rtph264pay config-interval=1 ! udpsink host=mycomputer.local port=5000`

**computer:** `gst-launch-1.0 udpsrc port=5000 caps="application/x-rtp,media=(string)video,clock-rate=(int)90000,encoding-name=(string)H264" ! rtph264depay ! decodebin ! autovideosink`

Works all fine and with super low latency.

--

However, I would like to do this via **TCP**, mainly because I want the Pi to be a server that clients can connect to (the Pi does not know ahead of time the IPs of the clients it shall send the video to).

Could you make an example how to do this?

I've searched on this matter for aroun 10 hours so far, but the only way I've found it to work was to use `gdppay`, which is a non-standard gstreamer protocol that other applications are unlikely to speak.

Thanks, and thanks for `rpicamsrc`!

nh2, 2015-02-27:
I have found this thread that seems to go into the same direction, but I don't fully understand it and it did not give a solution that worked for me: http://gstreamer-devel.966125.n4.nabble.com/h-264-Video-Streaming-over-TCP-td4658747.html

nh2, 2015-03-03:
As a current workaround, I use the `gst-rtsp-server` project, running [`test-launch.c`](http://cgit.freedesktop.org/gstreamer/gst-rtsp-server/plain/examples/test-launch.c) with this command line:

```shell
./test-launch "( rpicamsrc bitrate=1000000 ! video/x-h264,width=1280,height=720,framerate=30/1 ! rtph264pay config-interval=1 name=pay0 pt=96 )"
```

That works pretty sweet, and I can use as clients:
* **VLC:** `vlc rtsp://192.168.178.42:8554/test`
* But **gstreamer** has lower latency: `gst-launch-1.0 rtspsrc location=rtsp://myraspiIP:8554/test latency=0 ! rtph264depay ! decodebin ! autovideosink`

thaytan, 2015-03-04:
The problem is you need some sort of encapsulation that preserves H.264 packet boundaries and packet timestamps, which is streamable. There's a few options for that:

* RTP, as you've already discovered, which can work over TCP and gst-rtsp-server can do that. You can use rtspt:// to force a TCP connection, or it'll try UDP and then fall back to TCP after a few seconds if UDP isn't working.
* A streamable container, such as MPEG-TS or Matroska. For example:
gst-launch-1.0 rpicamsrc ! h264parse config-interval ! mpegtsmux ! tcpserversink -v
and then on the receiver:
gst-launch-1.0 tcpclientsrc ! decodebin ! videoconvert ! autovideosink
or you can use 'matroskamux streamable=true' in place of mpegtsmux.

For both VLC and GStreamer with RTSP, you have some control over the latency. There's a 'network buffer size' setting for VLC somewhere, and a 'latency' property on GStreamer's rtspsrc that's 2000ms (2 seconds) by default.

thaytan, 2015-03-06:
Please reopen if you need any more info

closed 2015-03-06 by thaytan

towolf, 2015-05-07:
I’d have a question and I’m pretty sure you’d know ...

I’m fond of RTSP server option, because that spawns the pipeline on demand and supports multiple clients with "shared" factory. The camera doesn’t needless run.

I made this simple RTSP server in Python. And I wonder if there is any way to access the `rpicamsrc` *somehow* while it is launched to change the annotation text.

I suspect the pipeline is not accessible, but I’m too much of a noob to understand what is possible.

```python
#!/usr/bin/python3

import sys
import gi
gi.require_version('Gst', '1.0')
from gi.repository import GObject, Gst, GstRtspServer

if __name__ == "__main__":
    GObject.threads_init()
    loop = GObject.MainLoop()
    Gst.init(None)

    server = GstRtspServer.RTSPServer()
    server.set_service('554')

    factory = GstRtspServer.RTSPMediaFactory()
    factory.set_launch("( rpicamsrc name=src bitrate=0 quantisation-parameter=24 intra-refresh-type=cyclic metering-mode=matrix annotation-mode=12 preview=false ! video/x-h264,profile=baseline,width=640,height=480,framerate=30/1 ! h264parse ! rtph264pay pt=96 name=pay0 )")
    factory.set_shared(True)

    server.get_mount_points().add_factory("/video", factory)
    
    server.attach(None)
    print("stream ready at rtsp://paitu:554/video") 

    loop.run()

```

thaytan, 2015-05-07:
You should be able to do what you want via the 'media-configure' signal on the factory, called when the pipeline is created.

You'll need to translate to python from C, but the test-appsrc example in the gst-rtsp-server examples does something similar enough:

http://cgit.freedesktop.org/gstreamer/gst-rtsp-server/tree/examples/test-appsrc.c

towolf, 2015-05-07:
Thanks a lot. That was what I needed.

I got a little further, but I wonder how to handle a periodic GObject timeout.

```python
#!/usr/bin/python3

import gi
gi.require_version('Gst', '1.0')
from gi.repository import GObject, Gst, GstRtspServer
import time

def on_media_configure(factory, media):
    print("configured")

    cam = media.get_element().get_by_name('src')
    GObject.timeout_add(1/30 * 1000, update_annotation, cam)

def update_annotation(cam):
    try:
        cam.set_property("annotation_text", "%9.3f\n" % time.time());
    except Exception as e:
        print(e)
        return False
    return True
        
if __name__ == "__main__":
    GObject.threads_init()
    loop = GObject.MainLoop()
    Gst.init(None)

    server = GstRtspServer.RTSPServer()
    server.set_service('554')

    factory = GstRtspServer.RTSPMediaFactory()
    factory.set_launch("( rpicamsrc name=src bitrate=0 quantisation-parameter=24 intra-refresh-type=cyclic metering-mode=matrix annotation-mode=13 preview=false  ! video/x-h264,profile=high,width=640,height=480,framerate=30/1 ! h264parse ! rtph264pay pt=96 name=pay0 )")
    factory.set_shared(True)
    factory.connect("media-configure", on_media_configure)

    server.get_mount_points().add_factory("/video", factory)
    
    server.attach(None)
    print("stream ready at rtsp://paitu:554/video")

    loop.run()
```

How do I handle `GObject.timeout_add()` (more) properly here? Is the `return False` enough to clear it when the pipeline goes down? Somehow that exception is never reached.

thaytan, 2015-05-07:
Once Python is holding a reference to the source element, it will never be destroyed - so your set_property will continue to succeed even after the RTSP server removes the media.

In the media-configure, you can attach to the 'unprepare' signal on the new media item, and detach your callback and drop your reference to the source element then.

See http://gstreamer.freedesktop.org/data/doc/gstreamer/head/gst-rtsp-server/html/GstRTSPMedia.html#GstRTSPMedia-unprepared

and

http://cgit.freedesktop.org/gstreamer/gst-rtsp-server/tree/gst/rtsp-server/rtsp-media.c#n55

towolf, 2015-05-07:
Thanks again.

I added `on_unprepared` callback. And it clears the timer. But I’m concerned about leaking the rpicam elements as you explained. And why are they leaking if the reference is overwritten? Are they not getting garbage collected?. To "drop" my reference can’t possibly mean just `del(cam)`? Not so sure about scope and the interaction with Gstreamer.

```python
def on_unprepared(media, timeout, cam):
    print("de-configured")
    GObject.source_remove(timeout)
    del(cam)

def on_media_configure(factory, media):
    print("configured")

    cam = media.get_element().get_by_name('src')
    timeout = GObject.timeout_add(1/30 * 1000, update_annotation, cam)

    media.connect("unprepared", on_unprepared, timeout, cam)
```

An additional question, perhaps? It may be worth to open another issue number.

In the original raspicam one can set `--fps 0` to give variable framerates. In night mode the FPS goes down during the day it ramps up (up to 30 by default). This has certain advantages during day and night operation.

Would this be compatible with rpicamsrc? And with RTSP? Dunno how that would would with setting caps on rpicamsrc ...

thaytan, 2015-05-08:
On 08/05/15 04:09, Tobias Wolf wrote:
> Thanks again.
>
> I added |on_unprepared| callback. And it clears the timer. But I’m
> concerned about leaking the rpicam elements as you explained. And why
> are they leaking if the reference is [...]

Yes, that's all that's required - once the Python variable is cleared 
and garbage collected, it will drop the reference it's holding at that 
GStreamer level, allowing the camera source to be freed.

> def  on_unprepared(media,timeout,cam):
>      print("de-configured")
>      GObject.source_remove(timeout)
>      del(cam)
>
>
> def  on_media_configure(factory,media):
>      print("configured")
>
>      cam= [...]

Looks right.

> An additional question, perhaps? It may be worth to open another issue
> number.
>
> In the original raspicam one can set |--fps 0| to give variable
> framerates. In night mode the FPS goes down during the [...]

If you set a framerate of 0/1 it should enable variable fps, with 
maximum framerate limited by the sensor mode / resolution you're 
capturing. I implemented that, but haven't tested - let me know how it goes.

towolf, 2015-05-09:
> If you set a framerate of 0/1 it should enable variable fps, with maximum framerate limited by the sensor mode / resolution you're capturing. I implemented that, but haven't tested - let me know how it goes.

This works very nicely. In combination with exposure-mode=night it ramps FPS down to 1.3 to boost dark scenes. Of course anything moving will be blurry. It’s a question of what shutter speeds one needs (still wondering what the best (easy) way would be to control these properties from outside of the process, I used to use `v4l2-ctl` utility)

Now, I thought about opening the port to the internet. That would at least requires basic auth, I guess.

Maybe my choice of using python was wrong because I cannot translate the example auth code from C to python.

Looking at test-video.c I need to make a special token and add it to an auth object.

```c
auth = gst_rtsp_auth_new ();
[...]
 /* make user token */
  token =
      gst_rtsp_token_new (GST_RTSP_TOKEN_MEDIA_FACTORY_ROLE, G_TYPE_STRING,
      "user", NULL);
  basic = gst_rtsp_auth_make_basic ("user", "password");
  gst_rtsp_auth_add_basic (auth, basic, token);
[...]
  /* configure in the server */
  gst_rtsp_server_set_auth (server, auth);

[and later]

  permissions = gst_rtsp_permissions_new ();
  gst_rtsp_permissions_add_role (permissions, "user",
      GST_RTSP_PERM_MEDIA_FACTORY_ACCESS, G_TYPE_BOOLEAN, TRUE,
      GST_RTSP_PERM_MEDIA_FACTORY_CONSTRUCT, G_TYPE_BOOLEAN, TRUE, NULL);
  gst_rtsp_media_factory_set_permissions (factory, permissions);

```

Something like

```python
auth = GstRtspServer.RTSPAuth()
token = GstRtspServer.RTSPToken(GstRtspServer.RTSP_TOKEN_MEDIA_FACTORY_ROLE, 
                   GObject.TYPE_STRING, 
                   'user', 
                   None)

# But this fails with: 
# TypeError: Passing arguments to gi.types.Boxed.__init__() is deprecated. All arguments passed will be ignored.

basic = auth.make_basic("user", "password")
auth.add_basic(basic, token)
server.set_auth(auth)

[and later]

permissions = GstRtspServer.RTSPPermissions()

# but neither permissions nor auth have a method add_role(), which I need
```

Is this a problem with my python-gobject (I have 3.16.1) or gobject-introspection (1.44.0-1)?

thaytan, 2015-05-09:
It may be a problem with the g-i markup for gst-rtsp-server - I'd need to dig through it and see.

towolf, 2015-05-09:
I tried digging, but with my almost zero knowledge this is labyrinthine.

Merely found this:
http://cgit.freedesktop.org/gstreamer/gst-python/commit/?id=4c08c3396c3cdadc939a9889f1e4664f938921a0

And things like `help(type(GstRtspServer.RTSPToken))` or the online docs
are really not so helpful when stuck somewhere.

Maybe I’ll not do authentication then.

cbasavaraj, 2019-11-29:
Hi, many years later, but I am using try `rtph264pay` over tcp myself, and found that `rtpstreampay` works well. So @nh2 could / could've used it instead of `gdppay`. Cheers

nh2, 2019-11-29:
@cbasavaraj Thanks! Do you have a full example pipeline handy by any chance that we could try out?

cbasavaraj, 2019-11-29:
Sure. I am using `picamera` to record directly to gstreamer in h264 format. (Note: I am using a Compute Module which allows two cameras, hence `stereo` mode.) On the desktop, I read using OpenCV so I can use for further processing. 

Sender (Pi):
```
    gstring = ["gst-launch-1.0 fdsrc",
               "h264parse",
               "rtph264pay config-interval=1 pt=96",
               "rtpstreampay",
               "tcpclientsink host=%s port=5000" % HOST_IP]

    gstring = " ! ".join(gstring)

    try:

        print('Starting PiCamera...')
        camera = picamera.PiCamera(stereo_mode='side-by-side',
                                   stereo_decimate=False,
                                   resolution=RES,
                                   framerate=FPS)

        time.sleep(2)

        print('Starting gstreamer...')
        print(gstring)
        gstreamer = subprocess.Popen(
                    gstring,
                    stdin=subprocess.PIPE,
                    shell=True)

        camera.start_recording(gstreamer.stdin, format='h264')
        camera.wait_recording(DURATION)
        camera.stop_recording()

    finally:

        camera.close()

        gstreamer.stdin.close()
        gstreamer.wait()
```

Receiver (Desktop):
```
    gstring = "tcpserversrc host=192.168.1.14 port=5000 ! application/x-rtp-stream,encoding-name=H264,payload=96 ! " + \
              "rtpstreamdepay ! rtph264depay ! avdec_h264 ! videoconvert ! appsink"

    recv_cap = cv2.VideoCapture(gstring, cv2.CAP_GSTREAMER)
```

thaytan, 2019-11-30:
rtpstreampay is also a non-standard GStreamer encapsulation though. Other clients can't play that

cbasavaraj, 2019-11-30:
Ah ok, I am fairly new to GStreamer, didn't know. I had some success doing TCP by using `matroskamux` (after `x264enc`) instead of `rtph264pay` + `rtpstreampay` , but it wan't working after `h264parse`. Don't know if `matroskamux` is something that can be read by other clients?

thaytan, 2019-12-03:
Yes, matroska is the container format used for webm video and supported by lots of players. Using that or MPEG-TS works as a streaming encapsulation for TCP too (for the receiver it's basically the same as playing a file over HTTP then)

cbasavaraj, 2019-12-03:
Ok thanks. We are using gstreamer internally only, so `rtph264pay` + `rtpstreampay` is ok for us. But will keep this in mind, and maybe try `matroskamux` again.