thaytan/gst-rpicamsrc #21: Variable bitrate support

nh2, 2015-03-01:
`raspivid` supports a variable bit rate, setting the `bitrate` to 0 and passing a quantisation parameter, e.g. `-qt 30`:

https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L149

This allows for high quality images on moved scenes and at the same time super low bandwidth use on still scenes.

For example, I get ~5KB/s pointing the camera into my room with VBR, and ~200KB/s when moving in front of it, using `raspivid -t 0 -w 1280 -h 720 -fps 30 -g 300 -b 0 -qp 30 -o - | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=llw port=5000`.

The related code where the `quantisationParameter` is passed to MMAL in `raspivid` seems to be here: https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L1514

Could we have variable bitrate support in `rpicamsrc` as well?

thaytan, 2015-03-05:
VBR should now work - please test, and reopen if you find anything wrong.

closed 2015-03-05 by thaytan

nh2, 2015-03-05:
Works really well, thanks a lot!