thaytan/gst-rpicamsrc #22: Property `intra-refresh-type` set to `cyclic` or `both`

towolf, 2015-03-18:
I’m kinda sure this worked before. But now when I set `intra-refresh-type=cyclic` or `intra-refresh-type=both` it bombs out.

```
mmal: mmal_vc_port_enable: failed to enable port vc.ril.video_encode:in:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable connected port (vc.ril.video_encode:in:0(OPQV))0x72104240 (EINVAL)
mmal: mmal_connection_enable: output port couldn't be enabled
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data flow error.
```

thaytan, 2015-03-24:
Both cyclic and both are working OK for me. Have you tried rpi-update and a reboot?

thaytan, 2015-04-09:
Any update for me?

towolf, 2015-04-09:
Sorry, missed your previous comment. I’m using Archlinux ARM. Firmware is `raspberrypi-firmware-20150406-1`. I’ve rebooted.

```
root@paitu:~# gst-launch-1.0 rpicamsrc bitrate=5000000 inline-headers=true intra-refresh-type=cyclic metering-mode=matrix annotation-mode=0x00000300 preview=false  ! 'video/x-h264,profile=baseline,width=1024,height=768,framerate=30/1' ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=10.0.0.9 port=9000 sync=false async=true
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mmal: mmal_vc_port_enable: failed to enable port vc.ril.video_encode:in:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable connected port (vc.ril.video_encode:in:0(OPQV))0x72004240 (EINVAL)
mmal: mmal_connection_enable: output port couldn't be enabled
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.199651609
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
root@paitu:~# gst-launch-1.0 rpicamsrc bitrate=5000000 inline-headers=true intra-refresh-type=both metering-mode=matrix annotation-mode=0x00000300 preview=false  ! 'video/x-h264,profile=baseline,width=1024,height=768,framerate=30/1' ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=10.0.0.9 port=9000 sync=false async=true
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mmal: mmal_vc_port_enable: failed to enable port vc.ril.video_encode:in:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable connected port (vc.ril.video_encode:in:0(OPQV))0x72104240 (EINVAL)
mmal: mmal_connection_enable: output port couldn't be enabled
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.196840156
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
root@paitu:~# gst-launch-1.0 rpicamsrc bitrate=5000000 inline-headers=true intra-refresh-type=cyclic-rows metering-mode=matrix annotation-mode=0x00000300 preview=false  ! 'video/x-h264,profile=baseline,width=1024,height=768,framerate=30/1' ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=10.0.0.9 port=9000 sync=false async=true
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:02.998113228
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

Will rebuild gst-rpicam from make distclean now. Dunno if that’s gonna make a difference.

towolf, 2015-04-09:
Rebuild the module. Even with simple pipeline it errors out with same error.

`gst-launch-1.0 rpicamsrc intra-refresh-type=cyclic ! fakesink`

thaytan, 2015-04-17:
Your test pipeline above failed for me this morning too. While I was trying to strip it down to compare against the pipeline that worked before, it suddenly started working. Now the full pipeline works again. I think the initialisation is racy. I'll try and see if it's in gst-rpicamsrc, or if it's a race inside the DSP init. The latter seems likely, since it's linked to intra-refresh-cyclic - which is just another property as far as gst-rpicamsrc is concerned.

Try running this and see if it ever eventually works for you:

while true; do gst-launch-1.0 rpicamsrc bitrate=5000000 inline-headers=true intra-refresh-type=cyclic-rows metering-mode=matrix annotation-mode=0x00000300 preview=false  ! 'video/x-h264,profile=baseline,width=1024,height=768,framerate=30/1' ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=10.0.0.9 port=9000 sync=false async=true; done

towolf, 2015-04-20:
Well, `cyclic-rows` always works. You meant I should test `cyclic` or `both`, I suppose?

I ran your loop with `cyclic` for a couple minutes. First try after 577 iterations/tries it stuck and ran. Second try it filled my (8192 rows) terminal scrollback and I canceled. Third try I booted fresh and inserted `echo` with iteration number. Canceled this at 5000 iterations. Then tried `both`, which worked after 1332 iterations

This is on Rpi 2. Could it be a multi-core/threading issue?

thaytan, 2015-04-20:
Ah, oops - I pasted the wrong line. Yes - `cyclic-rows` always works for me too, which is funny because it was the one value I'd mapped incorrectly and was setting a bogus value to MMAL. With that fixed, it now regularly fails as well. I guess mmal was ignoring the bogus value.

I'm testing on an RPi1 - so not an RPi2 specific issue.

I wonder if it's an uninitialised-memory use issue - if it happens to be zero or so it works. I can't install valgrind (seems to be a raspbian repo issue) without compiling it first to see if there's an uninited access anywhere in userspace.

towolf, 2015-04-20:
> Yes - `cyclic-rows` always works for me too, which is funny because it was the one value I'd mapped incorrectly and was setting a bogus value to MMAL.

Aha. That’s why I was trying the other one. Because this one didn’t seem to have any effect.

>  I can't install valgrind (seems to be a raspbian repo issue) without compiling it first to see if there's an uninited access anywhere in userspace.

Arch has valgrind. Do you want me to run something and paste the output?

```
root@paitu:~# pacman -Ss valgrind
extra/valgrind 3.10.1-1
    A tool to help find memory-management problems in programs
```

thaytan, 2015-04-20:
Oh, yes please -

G_SLICE=always-malloc G_DEBUG=gc-friendly valgrind --leak-resolution=high --leak-check=yes --num-callers=20 --alignment=16 gst-launch-1.0 rpicamsrc intra-refresh-type=both num-buffers=5 ! fakesink

might do the trick.

towolf, 2015-04-20:
It’s sitting there spinning CPU3. Gotto go to garden center to buy some plants. Will update if it finishes by the time I get back.

```
root@paitu:~# G_SLICE=always-malloc G_DEBUG=gc-friendly valgrind --leak-resolution=high --leak-check=yes --num-callers=20 --alignment=16 gst-launch-1.0 rpicamsrc intra-refresh-type=both num-buffers=5 ! fakesink
==9551== Memcheck, a memory error detector
==9551== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==9551== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==9551== Command: gst-launch-1.0 rpicamsrc intra-refresh-type=both num-buffers=5 ! fakesink
==9551== 
==9551== Invalid read of size 4
==9551==    at 0x4019974: index (in /usr/lib/ld-2.20.so)
==9551==    by 0x40148DB: dl_open_worker (in /usr/lib/ld-2.20.so)
==9551==  Address 0x5102e58 is 40 bytes inside a block of size 42 alloc'd
==9551==    at 0x4834780: malloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5319E14: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5319E20: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5319E28: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Warning: noted but unhandled ioctl 0xc410 with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==9551== Warning: noted but unhandled ioctl 0xc400 with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==9551== Warning: noted but unhandled ioctl 0xc40d with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==9551== Warning: noted but unhandled ioctl 0xc40c with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==9551== Thread 2 VCHIQ completio:
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A060: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A110: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A124: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A12C: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x531A090: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Thread 4 HTV Notify:
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x527AB38: tvservice_notify_func (in /opt/vc/lib/libbcm_host.so)
==9551== 
Setting pipeline to PAUSED ...
==9551== Thread 2 VCHIQ completio:
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A07C: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5254DC8: mmal_vc_vchiq_callback (in /opt/vc/lib/libmmal_vc_client.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x5254DC8: mmal_vc_vchiq_callback (in /opt/vc/lib/libmmal_vc_client.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x5254F30: mmal_vc_vchiq_callback (in /opt/vc/lib/libmmal_vc_client.so)
==9551== 
[... 731 lines cut ...]
==9551==    at 0x522CF00: mmal_component_destroy (in /opt/vc/lib/libmmal_core.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5258060: mmal_vc_component_destroy (in /opt/vc/lib/libmmal_vc_client.so)
==9551== 
==9551== Warning: noted but unhandled ioctl 0xc40b with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==9551== Thread 2 VCHIQ completio:
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x5254DC0: mmal_vc_vchiq_callback (in /opt/vc/lib/libmmal_vc_client.so)
==9551== 
==9551== Conditional jump or move depends on uninitialised value(s)
==9551==    at 0x531A09C: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Use of uninitialised value of size 4
==9551==    at 0x531A0B4: completion_thread (in /opt/vc/lib/libvchiq_arm.so)
==9551== 
==9551== Warning: noted but unhandled ioctl 0xc411 with no size/direction hints.
==9551==    This could cause spurious value errors to appear.
==9551==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
Setting pipeline to NULL ...
Freeing pipeline ...

```

thaytan, 2015-04-20:
Thanks for that. I don't see anything obvious in there, although there's clearly a lot of incompletely initialised memory usage inside mmal and friends. 

I did just notice that if I turn on some GStreamer debug levels, I can make it work reliably. Either debug level slow things down enough that it seems to init reliably, or it changes the memory around in some way. Even when not printing anything, just skipping ignored internal debug lines is enough:

GST_DEBUG=identity:6 gst-launch-1.0 rpicamsrc intra-refresh-type=both num-buffers=5 ! fakesink

works reliably, where

gst-launch-1.0 rpicamsrc intra-refresh-type=both num-buffers=5 ! fakesink

doesn't. It's also interesting that I can reproduce it reliably now, where I couldn't when you first reported the bug. I wonder if something in the firmware changed.

towolf, 2015-04-20:
Just came home, it was still spinning the CPU. When I canceled it printed this

```
^[[F^[[B^C==9551== 
==9551== HEAP SUMMARY:
==9551==     in use at exit: 272,215 bytes in 2,042 blocks
==9551==   total heap usage: 24,048 allocs, 22,006 frees, 1,927,635 bytes allocated
==9551== 
==9551== Thread 1:
==9551== 504 bytes in 39 blocks are possibly lost in loss record 1 of 9
==9551==    at 0x4834780: malloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==9551== 
==9551== 540 bytes in 5 blocks are possibly lost in loss record 2 of 9
==9551==    at 0x48370D4: realloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==9551== 
==9551== 4,096 bytes in 1 blocks are definitely lost in loss record 4 of 9
==9551==    at 0x4834780: malloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==9551== 
==9551== 4,342 bytes in 109 blocks are possibly lost in loss record 6 of 9
==9551==    at 0x4836E94: calloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==9551== 
==9551== LEAK SUMMARY:
==9551==    definitely lost: 4,096 bytes in 1 blocks
==9551==    indirectly lost: 0 bytes in 0 blocks
==9551==      possibly lost: 5,386 bytes in 153 blocks
==9551==    still reachable: 255,909 bytes in 1,809 blocks
==9551==         suppressed: 0 bytes in 0 blocks
==9551== Reachable blocks (those to which a pointer was found) are not shown.
==9551== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==9551== 
==9551== For counts of detected and suppressed errors, rerun with: -v
==9551== Use --track-origins=yes to see where uninitialised values come from
==9551== ERROR SUMMARY: 11402 errors from 246 contexts (suppressed: 3 from 1)

```

towolf, 2015-04-20:
From my PoV you don’t need to dive too deep into this, if it takes too much time.

I merely wanted to try these flags to see if they help with live streaming error resilience and/or latency.

thaytan, 2015-04-20:
if you could test with GST_DEBUG=identity:5 and see if that makes it work for you too, that'd be cool

towolf, 2015-04-20:
On Mon, Apr 20, 2015 at 6:15 PM, Jan Schmidt <notifications@github.com>
wrote:

> GST_DEBUG=identity:5

Yes, prepending that, it works. Funny.

thaytan, 2015-04-20:
I think I found it - that commit makes it work for me.

closed 2015-04-20 by thaytan

towolf, 2015-04-20:
Can confirm. Thank you very much.

commit 00010fab (2015-04-20): Clear intra-refresh MMAL param struct.

Use memset on the stack allocated MMAL_PARAMETER_VIDEO_INTRA_REFRESH_T
struct. Apparently mmal_port_parameter_get() doesn't retrieve all
parameters, causing random failures when we set the intra-refresh
param on the encoder.

Fixes #22 for me.