thaytan/gst-rpicamsrc #23: Can only update annotation text once per second.

ali1234, 2015-04-21:
This isn't fast enough! I can update it almost every frame with python-picamera.

My use case is displaying the camera's physical orientation in the video feed, as determined by an accelerometer and compass module, before sending the encoded video to gst-rtsp-server.

See https://github.com/ali1234/piroverd/blob/master/main.c#L39

thaytan, 2015-04-21:
The annotation updating code literally updates once per second, so that's not surprising. There's no reason though - I just copied the code over from raspivid.

There's no locking around property updates, which I need to add too - there's a small risk that changing the annotation on the fly could crash atm.

In the meantime, you could change the divisor on line 910 and 913 or RaspiCapture.c from 1000 to 40, to get 25 updates per second.

See https://github.com/thaytan/gst-rpicamsrc/blob/master/src/RaspiCapture.c#L910

thaytan, 2015-04-26:
Current git master applies annotation (and other properties, where possible) updates on the fly, and should do what you want.

closed 2015-04-26 by thaytan

thaytan, 2015-04-27:
Oh, and in your pirover - you no longer need to set do-timestamp=true on the rpicamsrc. It will internally timestamp using the capture timestamps. do-timestamp will overwrite those with more jittery versions based on the time GStreamer receives them.

ali1234, 2015-04-27:
Annotations don't work at all after 9ab490 "Merge changes from userland repo". I am using stock raspbian kernel/firmware/userland. Is this expected or do I need to change my code?

thaytan, 2015-04-27:
No, that's not expected. This pipeline works (and produces annotation text at the top) for me, with git master and stock kernel/firmware/userland, on RPI1:

gst-launch-1.0 -e rpicamsrc annotation-mode=custom-text annotation-text="Test Annotation" ! video/x-h264,width=640,height=480 ! h264parse ! mp4mux ! filesink location=test.mp4

ali1234, 2015-04-27:
Sorted it. It is now necessary to specify annotation-text when creating the element. Previously you could leave it out of the launch string and then set the value later, but this no longer seems to work.

thaytan, 2015-04-27:
That's weird - I'm not sure why that would happen

ali1234, 2015-04-27:
Neither am I. Something weird is going on here. Now the resulting video stream is always 1920x1080 regardless of what width and height the pipeline asks for.

ali1234, 2015-04-27:
Right> I double checked everything and this is what I found. Using the example you gave:

gst-launch-1.0 -e rpicamsrc annotation-mode=custom-text annotation-text="Test Annotation" ! video/x-h264,width=640,height=480 ! h264parse ! mp4mux ! filesink location=test.mp4

Using a stock, fully up to date raspbian, this produces no annotation in the output video at all, regardless of whether you set it in the launch string or via the property.

However, after running rpi-update to get the newest firmware, annotation once again works exactly as expected. It is not necessary to specify annotation-text, you can just set the property whenever, as before.

thaytan, 2015-04-27:
Ah, I see. They change to a new annotation API in newer firmware, and deprecated the old. I updated rpicamsrc to use the newer API - as raspivid does.

I just pushed a new commit that fixes the silly mistake around configuring capture size. That should work again now.

ali1234, 2015-04-27:
Everything is working okay now with rpi-update firmware. Thanks.

thaytan, 2015-04-27:
Cheers, thanks for testing :)