thaytan/gst-rpicamsrc #26: AWB settings are ignored

shrx, 2015-05-05:
Setting `awb-mode` does not have any effect on the image, also setting `awb-mode=off` and manually `awb-gain-blue=4.2 awb-gain-red=1.0` doesn't have any effect, the image is always slightly too red.

shrx, 2015-05-05:
Ok, this is only an issue if I also set `preview=false`. Why is that?

shrx, 2015-05-05:
Could it be related to this discrepancy in the documentation? `awb-mode` is set to `off` (lowercase), but documentation says
> `awb-gain-blue       : Manual AWB Gain for blue channel when awb-mode=OFF` (uppercase)

thaytan, 2015-05-11:
I updated the doc strings (thanks!), but I'm not sure why the properties wouldn't have an effect when awb-mode=off, or why changing the preview would make a difference. I'll try and remember to test when I'm physically in front of my rpi

towolf, 2015-05-16:
Sounds like [this is the bug in question](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=110273&p=758229&hilit=awb)?

shrx, 2015-05-16:
Yep, looks like it is.

thaytan, 2015-10-10:
There was a silly bug in my code setting the causing both awb-red and awb-blue to affect only the red channel. It's fixed by the commit above, so things might work better now.

shrx, 2015-10-12:
Great, nice to see this fixed. I don't have the time to test it, but will reopen if anyone still has an issue with this.

closed 2015-10-12 by shrx