thaytan/gst-rpicamsrc #32: Could you clarify whether this used HW encoding?

samskiter, 2015-07-07:
Sorry for the very basic question, but I'm not particularly familiar with video encoding in general. Does this module make use of the hardware video encoding on the Pi or does it do a manual encode?

Thanks a lot for this great module :)

thaytan, 2015-07-10:
Yes, it uses the hardware encoder on the raspberry pi GPU

closed 2015-07-10 by thaytan

kaihendry, 2015-07-15:
Sorry for the stupid question here, but I have a simple use case where I need to re-encode MOV files taken from another old Webcam to MP4, so I can play them back from my iPhone's Safari. I would like to use my Rpi2s hardware accelerated MP4 encoding to do this.

I think your tool is what I need, but I don't have a good idea how to build the pipeline. Could you offer some simple usage suggestions please?