thaytan/gst-rpicamsrc #33: Latency

EmbeddedSam, 2015-07-15:
What kind of latency could one expect from this module from camera to display <100ms? I am really looking for something which is around 720p with <10ms latency but I think this might be pi in the sky.

philn, 2015-07-15:
See also #31

thaytan, 2015-07-18:
The maximum framerate at 720p is 49fps, which gives a minimum frame duration of around 20ms, but that may not be the lower bound on latency - I'm not sure. The capture+encode may be able to happen faster or slower - depending on camera exposure time, or it may be fixed to that 20ms lower bound at 720p.

The encoder doesn't have a 'zerolatency' mode like x264 does - it only outputs completed encoded buffers, so the minimum latency does have a lower bound based on camera exposure and encoding time.

Beyond that, of course, there's whatever transmission and receiver delay. In my experience, most of the latency is in the transmission and whatever the receiver has configured as a receive buffer.

Based on a quick and possibly misleading test, regardless of framerate I am seeing a fairly consistent 40-50ms latency between the capture timestamp reported by MMAL and the time at which it hands over the encoded buffer. That value may be due to the exposure time of the camera in the scene I tested.

sonium0, 2016-04-02:
I have seen that there is a YUV version of raspivid (RaspiVidYUV). Would it be possible to reduce the latency by using the YUV stream and encode with zerolatency-x264 afterwards?