thaytan/gst-rpicamsrc #34: Autogen failing

Nickalus, 2015-07-28:
I've compiled all gstreamer libraries from source without a hitch. I've tried to install this plugin and I can't get past running autogen. Here is my output:
 ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
src/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
src/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:1:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1
autogen.sh failed

Any help would be awesome!

thaytan, 2015-08-02:
I don't know why that would fail if libtool is installed and the LT_INIT macro is available.

Which version of libtool do you have?

Nickalus, 2015-08-03:
This can be closed. It happened to be a stupid error on my part. Libtool was installed in /usr/local/bin, when it needed to be installed in the same directory as automake (/usr/bin). This issue can be closed. Thanks for the awesome work!

closed 2015-08-03 by Nickalus