thaytan/gst-rpicamsrc #37: autogen.sh fails on updated RPI

latkinso42, 2015-08-06:
1366  ./autogen.sh --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf/
 1367  README
 1368  more README
 1369  more REQUIREMENTS 
 1370  sudo apt-get install autoreconf
 1371  sudo apt-get install autoconf
 1372  sudo apt-get install automake
 1373  sudo apt-get install libtool
 1374  sudo apt-get install libgstreamer1.0-dev
 1375  sudo apt-get install libgstreamer-plugins-base1.0-dev
 1376  more REQUIREMENTS 
 1377  sudo apt-get install libraspberrypi-dev
 1378  nano configure.ac 
 1379  history
pi@picam001-f ~/gst-rpicamsrc $ !1366
./autogen.sh --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf/
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:23: error: possibly undefined macro: AC_SUBST
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:34: error: possibly undefined macro: AC_MSG_ERROR
autoreconf: /usr/bin/autoconf failed with exit status: 1
autogen.sh failed

thaytan, 2015-08-06:
It don't see pkg-config in your package list, it might be that. In general on Debian the easiest way to be sure is to run ' sudo apt-get build-dep libgstreamer1.0-dev' to get all GStreamer core's build dependencies.

Hopefully that fixes it. I'll add it to the REQUIREMENTS.

latkinso42, 2015-08-06:
I had to update the URIs in my sources.list to:

deb http://archive.raspbian.org/raspbian wheezy main contrib non-free rpi
deb http://mirrordirector.raspbian.org/raspbian wheezy main contrib non-free rpi
deb-src http://archive.raspbian.org/raspbian wheezy main contrib non-free rpi
deb-src http://mirrordirector.raspbian.org/raspbian wheezy main contrib non-free rpi
#
deb http://www.linux-projects.org/listing/uv4l_repo/raspbian/ wheezy main

which helped me avoid  the "E: You must put some 'source' URIs in your sources.list" error,
however,  now i get:

pi@picam001-f ~ $ sudo apt-get build-dep libgstreamer1.0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Picking 'gstreamer1.0' as source package instead of 'libgstreamer1.0-dev'
E: Unable to find a source package for gstreamer1.0

Notice the "Picking 'gstreamer1.0' as source package instead of 'libgstreamer1.0-dev'"
and yet I seem to already have it:

pi@picam001-f ~ $ apt-cache show -n libgstreamer1.0-dev
Package: libgstreamer1.0-dev
Source: gstreamer1.0
Version: 1.2.0-1rpi3
Architecture: armhf
Maintainer: Maintainers of GStreamer packages <pkg-gstreamer-maintainers@lists.alioth.debian.org>
Installed-Size: 4071
Depends: libgstreamer1.0-0 (= 1.2.0-1rpi3), libc6-dev | libc-dev, pkg-config, libglib2.0-dev, libc6 (>= 2.13-28), libglib2.0-0 (>= 2.32), gir1.2-gstreamer-1.0 (= 1.2.0-1rpi3)
Recommends: debhelper
Suggests: gstreamer1.0-doc
Homepage: http://gstreamer.freedesktop.org
Priority: optional
Section: libdevel
Filename: pool/main/g/gstreamer1.0/libgstreamer1.0-dev_1.2.0-1rpi3_armhf.deb
Size: 1232476
SHA256: 4769763a58accd82aea5d03d6c7f3b776a134844cd419f703e0572174df42257
SHA1: e4daaa8d3b3383df697baeee69dc5b3cdfe4b513
MD5sum: f7e63dda0aaa8e51f3fe3d7c216b72dc
Description: GStreamer core development files
 GStreamer is a streaming media framework, based on graphs of filters
 which operate on media data.  Applications using this library can do
 anything from real-time sound processing to playing videos, and just
 about anything else media-related.  Its plugin-based architecture means
 that new data types or processing capabilities can be added simply by
 installing new plug-ins.
 .
 This package contains development files for the core library and
 elements.

Not too sure what to do now.. its weird..

thaytan, 2015-08-07:
apt-cache show will show uninstalled packages, and doesn't relate to install the build-dep packages. The build-deps of libgstreamer1.0-dev are the packages that were needed to build it. I don't know why the gstreamer1.0 source package isn't being found in the src repos you added. Did you run 'apt-get update' after adding them?

Anyway, let's ignore all that and just try doing 'apt-get install pkg-config' directly and see if pkg-config is all that's missing.

thaytan, 2015-12-01:
Please re-open if you're still having trouble.

closed 2015-12-01 by thaytan

BloodyF, 2016-04-27:
Please help me! I have the same error as latkinso42 even though pkg-config installed. :((
I googlinged during these week and often the result is: install pkg-config, libtool .... 
but I have libtool, pkg-config, automake, autoconf ... installed. Some results are noticeable version incompatibility, autocof do not see pk-config, libtool ...
I also encountered similar problems when building GStreamer manual (currently I'm using the GStreamer from: sudo apt-get install gstreamer1.0)
I am very grateful for any help.

thaytan, 2016-04-27:
What output do you get when you run autogen.sh?

BloodyF, 2016-04-28:
Thanks for your enthusiasm. I got same output as latkinso42.
I've just installed the Jessi insteed of Wheezy and successfuly run 3 commands which you gave. 
Can you show me how to verify gst-rpicamsrc if successfully installed or not?