thaytan/gst-rpicamsrc #41: Package does not contain libs if local compilation has not been done

fthiery, 2015-12-16:
After patching gst-rpicamsrc/debian/rules with || true so that the find|xargs doesn't fail, the following will produce an empty package:

```
cd gst-rpicamsrc
dpkg-buildpackage
cd ../
sudo dpkg -i gstreamer1.0-rpicamsrc_1.0.0_armhf.deb
```
The package didn't install the libs at all:
```
dpkg -L  gstreamer1.0-rpicamsrc
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/gstreamer1.0-rpicamsrc
/usr/share/doc/gstreamer1.0-rpicamsrc/copyright
/usr/share/doc/gstreamer1.0-rpicamsrc/changelog.gz
```
However, if i do:
```
cd gst-rpicamsrc
./autogen.sh
make
dpkg-buildpackage
cd ../
sudo dpkg -i gstreamer1.0-rpicamsrc_1.0.0_armhf.deb
```

This time, the package does contain the libs !
```
$ dpkg -L gstreamer1.0-rpicamsrc
/.
/usr
/usr/lib
/usr/lib/arm-linux-gnueabihf
/usr/lib/arm-linux-gnueabihf/gstreamer-1.0
/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
/usr/share
/usr/share/doc
/usr/share/doc/gstreamer1.0-rpicamsrc
/usr/share/doc/gstreamer1.0-rpicamsrc/copyright
/usr/share/doc/gstreamer1.0-rpicamsrc/changelog.gz
```

Shouldn't dpkg-buildpackage compile on it's own ?

thaytan, 2015-12-17:
Yes, it should. Not sure I ever actually tested the contributed debian/ subdir. I think I did, but possibly not. If I did, I would have used 'debuild -rfakeroot'

thaytan, 2015-12-17:
Hmm - I can't reproduce the failure you report:

git clean -f -x -d
./autogen.sh 
dpkg-buildpackage 
dpkg -c ../gstreamer1.0-rpicamsrc_1.0.0_armhf.deb 

drwxr-xr-x root/root         0 2015-12-18 00:26 ./
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/lib/
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/lib/arm-linux-gnueabihf/
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/lib/arm-linux-gnueabihf/gstreamer-1.0/
-rw-r--r-- root/root     67716 2015-12-18 00:26 ./usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/share/
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/share/doc/
drwxr-xr-x root/root         0 2015-12-18 00:26 ./usr/share/doc/gstreamer1.0-rpicamsrc/
-rw-r--r-- root/root      2183 2015-05-11 18:22 ./usr/share/doc/gstreamer1.0-rpicamsrc/copyright
-rw-r--r-- root/root       149 2015-05-11 18:22 ./usr/share/doc/gstreamer1.0-rpicamsrc/changelog.gz

fthiery, 2015-12-20:
Can you try without running ./autogen.sh ? My understanding is that dpkg-buildpackage should be enough (and should run autogen), don't you ?

thaytan, 2015-12-20:
I would not expect dpkg-buildpackage to run autogen.sh - that's normally a manual step that you only run when checking out from version control directly, and comes under different names.

fthiery, 2015-12-22:
I agree, but then release tarballs should be made available (with the configure files generated with make dist), at least accordingly to http://stackoverflow.com/questions/31359171/git-buildpackage-doesnt-call-configure-why

So there are 3 options
* add to the doc that to generate debian packages, ./autogen.sh && dpkg-buildpackage should be run
* provide classic release tarballs with configure script
* have dpkg-buildpackage run autogen:

```
override_dh_auto_configure:
    ./autogen.sh
```

What do you think ?

I think this is why the build process was failing (and that's why i added || true) : i was not running autogen before dpkg-buildpackage, so that's why the la failes were never generated and failed to delete. Maybe it would be better to revert my pull request (it's somewhat better to fail building the package rathe than building an empty one... even if no message indicates why it fails).

mlankhorst, 2017-11-28:
Oh actually I have a fix for this in my pull request. :)