thaytan/gst-rpicamsrc #43: bug in gst video force key unit event

chetanbnaik, 2015-12-31:
**when I launch the following gstreamer pipeline,**
gst-launch-1.0 -v rpicamsrc bitrate=3000000 rotation=180 ! video/x-   h264,width=640,height=480,framerate=30/1 ! h264parse ! mpegtsmux ! hlssink

**hlssink will emit an upstream force key unit event. however, the event handling by "rpicamsrc" fails.**

(gst-launch-1.0:2462): GStreamer-CRITICAL **: gst_event_get_structure: assertion 'GST_IS_EVENT (event)' failed

(gst-launch-1.0:2462): GStreamer-CRITICAL **: gst_event_get_seqnum: assertion 'GST_IS_EVENT (event)' failed

(gst-launch-1.0:2462): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object->refcount > 0' failed

chetanbnaik, 2015-12-31:
on a similar line, how is the "force key unit" supported in rpicamsrc?

thaytan, 2016-01-02:
force-key-unit events are handled by requesting an i-frame from the hardware H.264 encoder.

thaytan, 2016-01-02:
Actually, that doesn't seem sufficient to fix it for some reason.

thaytan, 2016-01-03:
nm, yes it does - I had a stale .so in my plugin path.

closed 2016-01-03 by thaytan

chetanbnaik, 2016-01-03:
I pulled the repository again and installed it. but I still get the same error. may you share, what you did to make it working?

thaytan, 2016-01-03:
The only required change is the commit I pushed above. I guess you should check you've rebuilt gst-rpicamsrc properly. If so, which versions of the other pieces are you using? GStreamer core, base and bad modules.

chetanbnaik, 2016-01-03:
The core and base plugins are at 1.2.0 and the bad plugins are at 1.2.1. All this on RPi 2 model B.

what about yours? I will upgrade the gstreamer to that version

chetanbnaik, 2016-01-03:
its working now. I cleaned up the previous builts and its fine now.

Thanks a lot!

chetanbnaik, 2016-01-03:
For me, the problem is solved.

I was trying few more stuff, i think there is another bug, which I suspect is based on something similar.

gst-launch-1.0 rpicamsrc bitrate=1500000 rotation=180 num-buffers=200 keyframe-interval=50 ! video/x-h264,width=640,height=480,framerate=25/1 ! h264parse ! mpegtsmux ! multifilesink next-file="key-frame" location="test%05d.ts"

the above gstreamer command should create 4 files, however, it creates only 1.

thaytan, 2016-01-04:
I think the problem there is not related - if I look at the output of '... ! mpegtsmux ! fakesink silent=false' with gst-launch-1.0 -v, I see that it's not setting delta-unit flags properly on the outgoing buffers, but the output of h264parse looks fine to me. That's with GStreamer 1.4.4. With git master of GStreamer, it's working properly.

commit fe12cf13 (2016-01-02): basesrc event handlers should not unref

Don't unref the passed event when handling events via
the GstBaseSrc src pad event handler - basesrc does
the unref. That breaks handling of upstream
force-key-unit events by unreffing twice.

Fixes #43