thaytan/gst-rpicamsrc #44: Headers not installed when `make install`

samskiter, 2016-01-04:
The INSTALL readme says that the headers should be found in /usr/local/include but after cloning and running configure & sudo make install I cannot find any headers to develop against.

thaytan, 2016-01-04:
The INSTALL file (like pretty much every FOSS project out there) is the stock one autoconf creates, with generic instructions.

rpicamsrc only installs a GStreamer plugin .so. There's no headers - only the rpicamsrc GStreamer element, which you develop against like any other GStreamer element

samskiter, 2016-01-07:
@thaytan Ah, good point, thanks. I'd been mistakenly trying to keep a reference to a gstrpicamsrc, forgetting you can access properties through their string names.

closed 2016-01-07 by samskiter