thaytan/gst-rpicamsrc #45: custom text and time stamp together as an annotation?

chetanbnaik, 2016-01-10:
hi, is it possible to overlay a text that contains both custom text and time stamp together?

thaytan, 2016-01-11:
The annotation-mode property is a bitmask. You can set any of custom-text (0x1) and date (0x4) and/or time (0x8) simultaneously by or'ing the values, or setting a string like 'annotation-mode=custom-text+date+time'

The actual custom text is set in the annotation-text property.

chetanbnaik, 2016-01-11:
thanks for the answer, it works perfectly when using command line..
however, I am doing these things through python.
if I put rpisrc.set_property("annotation-mode","custom-text+date+time"), it fails to recognize. Any solution?

chetanbnaik, 2016-01-11:
thanks a lot. I am using gstreamer through python. so the following works:
rpisrc.set_property("annotation-mode",0x1 + 0x4 + 0x8)

closed 2016-01-11 by chetanbnaik