thaytan/gst-rpicamsrc #46: requesting an intra frame while recording

chetanbnaik, 2016-01-28:
Hello thaytan,

As an alternative, I am trying to wrap the python picamera module with gstreamer pipeline using the appsrc element. 

in picamera module, I can redirect the stream to a custom output that must have a "write" method.
```
class CamStreamer(object):
    def __init__(self,videoFormat,resolution,framerate):
        self.size = 0
        self.capstring = 'video/x-'+videoFormat+',width='+str(resolution[0]) \
                         +',height='+str(resolution[1])+',framerate=' \
                         +str(framerate)+'/1'
        self.playing = False
        self.paused = False
        self.make_pipeline()
        self.play_pipeline()

    def make_pipeline(self):
        CLI = [
                  'appsrc name="source" ! ',
                  'h264parse ! video/x-h264,stream-format=avc ! ',
                  'h264parse ! video/x-h264,stream-format=byte-stream ! ',
                  'mpegtsmux ! filesink name="sink"'
                  ]
        gcmd = "".join(CLI)
        self.pipeline = Gst.parse_launch(gcmd)
        self.filesink = self.pipeline.get_by_name("sink")
        self.filesink.set_property("location","custom.ts")
        self.appsrc = self.pipeline.get_by_name("source")
        self.appsrc.set_property("is-live",True)
        self.gstcaps = Gst.Caps.from_string(self.capstring)
        self.appsrc.set_property("caps",self.gstcaps)

    def play_pipeline(self):
        self.pipeline.set_state(Gst.State.PLAYING)

    def stop_pipeline(self):
        self.pipeline.set_state(Gst.State.READY)
               
    def write(self,s):
        gstbuff = Gst.Buffer.new_wrapped(s)
        ret = self.appsrc.emit("push-buffer",gstbuff)

    def flush(self):
        self.stop_pipeline()
```
and then I use the camera object as below:
```
class RPiCam(object):
    def __init__(self):
        self.resolution = (640,480)
        self.bitrate = 1000000
        self.format = "h264"
        self.framerate = 30
        self.rotation = 180
        
        self.camera = picamera.PiCamera()
        self.camera.resolution = self.resolution
        self.camera.framerate = self.framerate

    def startCam(self):
        self.camera.start_recording(CamStreamer(self.format,self.resolution,
                                                     self.framerate),format=self.format, bitrate=self.bitrate)

    def recordCam(self,duration):
        self.camera.wait_recording(duration)

    def stopCam(self):
        self.camera.stop_recording()

    def closeCam(self):
        self.camera.close()

```

I wish to request an i frame while the recording is in progress. I see the following lines of code in RaspiCapture.c which give me some hints on how to request a key frame against an external event. Can you help me how to implement this using python?

```
gboolean raspi_capture_request_i_frame(RASPIVID_STATE *state)
{
   MMAL_PORT_T *encoder_output = NULL;
   MMAL_STATUS_T status;
   MMAL_PARAMETER_BOOLEAN_T param = {{  MMAL_PARAMETER_VIDEO_REQUEST_I_FRAME, sizeof(param)}, 1};
   encoder_output = state->encoder_component->output[0];
   status = mmal_port_parameter_set(encoder_output, &param.hdr);
   if (status != MMAL_SUCCESS)
   {
      vcos_log_error("Unable to request I-frame");
      return FALSE;
   }
   return TRUE;
}
```

Thanks!

thaytan, 2016-01-29:
No, I have no idea about the picamera python module.

If you want to request a keyframe using the GStreamer rpicamsrc element, you can do that via a keyunit even sent into the pipeline with something like:

from gi.repository import Gst
from gi.repository import GstVideo
e = GstVideo.video_event_new_upstream_force_key_unit (Gst.CLOCK_TIME_NONE, True, 0)
pipeline.send_event(e)

chetanbnaik, 2016-02-01:
I have implemented the send_event successfully, but I was trying to get things around using picamera. 
Thanks for attempting, I have posted this question on other forums.

Thanks!

closed 2016-02-01 by chetanbnaik