thaytan/gst-rpicamsrc #48: Implementing video/x-raw support?

bensch128, 2016-04-04:
Hi, 
   I'm a noob to the rpi + gstreamer and I had some questions:
Is there amy plan to implement video/x-raw cap support in rpicamsrc?
I'm building a project where I need to extract several ROIs (up to 12 or so) from a single frame.
Since video/x-raw is needed, in your opinion, does it really matter performance-wise if v4l2src or rpicamsrc is used?

Thank you,
Ben

thaytan, 2016-04-09:
It's possible, but not on my timetable. I want to merge JPEG format support first, at least.

thaytan, 2016-10-13:
I just pushed support for raw video and JPEG capture, selected through caps negotiation:

gst-launch-1.0 rpicamsrc ! video/x-raw,format=RGB,width=640,height=480 ! fakesink.

closed 2016-10-19 by thaytan

bensch128, 2016-10-20:
Nice job, thanks for that. I'm sure someone will find this patch useful.

Sincerely
Ben