thaytan/gst-rpicamsrc #49: ./autogen.sh stops with an error

RAJensen, 2016-04-18:
Hello.
I would like to compile gst-rpicamsrc on RPi 3 Jessie.
I used the instructions but for some reason i can't figure out i'll end up with an error when execute: `./autogen.sh --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf/`
Returns:
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autom4te: cannot create autom4te.cache: No such file or directory
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
autogen.sh failed

And i'm kind of lost now, so can someone tell what when wrong and how to fix it?
G-streamer is already installed with Jessie and working. Also the compiler is shipped with Jessie.
Thank you for the help in advance.

closed 2016-04-18 by RAJensen

thaytan, 2016-04-19:
You closed it - I guess you worked it out? Mind saying what the fix was? File permissions?

RAJensen, 2016-04-19:
Yes, if i know exactly why it createt the error.
But as i did't find anyone else on the web, that had the same error, I 
concluded there might be something wrong at my side of the desk.
So as a last ressort, after downloading some more stuff, that maybe was 
missing, did a reboot, tried again with the same error and then I did 
power off the Pi, wait 5 seconds and startet it up again and followed 
the instruction and it worked.
I did experiment with some home brewed Python script earlyer and I might 
have createt some kind of deadlock without knowledge of it.
3 times before this, i did have trouble with autogen-scripts, on a Pi 1, 
Pi 2 and now Pi 3 but after completly power down, issues are gone.
Also I noticed that my Wifi did not behave as expectet after a reboot, 
but first after power off and so on.
And yes it might been a permission trouble, as at one point back in 
time, I also had the great pleasure of been denied acces to a USB-HDD 
that was mountet for my user and even as a root, it still gave me denied 
acces. Reboot = acces denied..! Power Down = behavior as expectet.

So confused, but happy as it finally succeded.
:-)

Den 19-04-2016 kl. 13:44 skrev Jan Schmidt:

bespiloy, 2020-04-16:
I have same problem and still no success after doing reboot. Anyone can help will be appreciated. thank you. Below is the error message.

pi@rpi3114:~/gst-rpicamsrc $ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autom4te: cannot create autom4te.cache: No such file or directory
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
autogen.sh failed
pi@rpi3114:~/gst-rpicamsrc $

bespiloy, 2020-04-16:
never mine. I fixed it by adding permission: I did the following:

cd ..
sudo chown -R pi:pi  gst-rpicamsrc
cd gst-rpicamsrc
./autogen.sh

It helped me after googling and reading the "https://superuser.com/questions/762631/why-i-obtain-this-error-message-when-i-try-to-compile-cgminer-autoreconf-error" link.

Regards.