thaytan/gst-rpicamsrc #51: mmal_component_create_core: could not find component 'vc.ril.camera'

Abu-Abdullah, 2016-06-27:
Hi,

I'm getting this error:
 mmal_component_create_core: could not find component 'vc.ril.camera'

env: 
ubuntu_mate 16.04
RPI 3

command:
./test-launch "( rpicamsrc bitrate=1000000 ! rtph264pay config-interval=1 name=pay0 pt=96 )"

thaytan, 2016-06-28:
I think this means you haven't enabled the camera in /boot/config.txt or else it's not connected correctly. Does it work in raspivid?

Abu-Abdullah, 2016-06-28:
yes, it is working normally

Abu-Abdullah, 2016-06-28:
i tried on Raspian (latest image) as well the following:
./test-launch "( rpicamsrc bitrate=1000000 ! rtph264pay config-interval=1 name=pay0 pt=96 )"

and it throws:
(lt-test-launch:1044): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_int: assertion 'gst_structure_has_field (structure, field_name)' failed

(lt-test-launch:1044): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_fraction: assertion 'gst_structure_has_field (structure, field_name)' failed

** (lt-test-launch:1044): WARNING **: Unknown profile string in rpicamsrc caps: constrained-baseline
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x7302b3a0 (EINVAL)

thaytan, 2016-06-29:
I can fix the fixation warnings to do something more sensible in that case, but what it's really complaining about is that you haven't put any caps filter in to tell rpicamsrc which width/height you want it to configure the camera for.

Try something like rpicamsrc bitrate=1000000 ! video/x-h264,width=800,height=600 ! rtph264pay config-interval=1 name=pay0 pt=96

Abu-Abdullah, 2016-06-29:
ok now the messages disappear but still im still not able to play the stream.

mplayer (on windows) gives me something like:
rtsp: read error.
librtsp: server responds: 'PRTSP/1.0 503 Service Unavailable'
STREAM_LIVE555, URL: rtsp://192.168.1.5:8554/test
Failed to get a SDP description from URL "rtsp://192.168.1.5:8554/test": 503 Service Unavailable

VLC gives me:
access_realrtsp warning: Cseq mismatch, got 1, assumed 0
access_realrtsp warning: only real/helix rtsp servers supported for now
core error: open of `rtsp://192.168.1.5:8554/test' failed

thaytan, 2016-06-29:
Try running test-launch with GST_DEBUG=3 and see if it spits out any errors / warnings when you try and connect.

In case it helps, I use a pipeline like this for streaming:

rpicamsrc bitrate=750000 intra-refresh-type=both drc=medium ! video/x-h264,profile=baseline,width=640,height=480 ! h264parse config-interval=1 ! rtph264pay name=pay0 pt=96

Abu-Abdullah, 2016-06-30:
much appreciated, your pipeline worked with me on raspian. few comments:

- it is only working on raspian, ubuntu mate still has the same issue i.e. i got this message:
mmal: mmal_component_create_core: could not find component 'vc.ril.camera'

camera is enabled and i can see it.

- in raspian, stream is only limited to one request/player i.e. if i start mplayer and vlc at the same time i got the following:
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC

thaytan, 2016-06-30:
> much appreciated, your pipeline worked with me on raspian. few comments:
> 
>   * it is only working on raspian, ubuntu mate still has the same issue
>     i.e. i got this message: mmal: mmal_component_create_core: could not
>     find component 'vc.ril.camera'

I don't know what that is. It sounds like the camera setup is broken
outside of rpicamsrc in Ubuntu MATE, but I don't know why:

> camera is enabled and i can see it.

That's surprising - because gst-rpicamsrc is a ported version of
raspivid. Perhaps they modified something and modified raspivid accordingly.

>   * in raspian, stream is only limited to one request/player i.e. if i
>     start mplayer and vlc at the same time i got the following: mmal:
>     mmal_vc_component_enable: failed to enable component: ENOSPC

You need to modify test-launch (or use your own small app) and be sure
to set gst_rtsp_media_factory_set_shared (factory, TRUE); (see some of
the other files in the examples dir that do that)

Abu-Abdullah, 2016-06-30:
thanks for your support. at least i have a working code in raspian. i will check how to modify test-launch

great support. much appreciated

closed 2016-06-30 by Abu-Abdullah

Abu-Abdullah, 2016-07-05:
maybe the issue with ubuntu mate is related to the same as of #28
`mmal: mmal_component_create_core: could not find component 'vc.ril.camera'`

seiga-k, 2017-06-02:
I got same problem in same environment (Rpi3, Ubuntu 16.04, Gstreamer 1.12).
Did you solve the problem ? @Abu-Abdullah  

On my Rpi3, raspivid work correctly. Also the Gstreamer works well with USB web cam.
Only rpicamsrc does'nt work.

I did ```ldd``` to libgstrpicamsrc.so and got the message as bellow. 
```
$ldd libgstrpicamsrc.so | grep mmal
        libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76c4c000)
        libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76c2b000)
```
It seems ```libmmal_core.so``` and ```libmmal_util.so``` are correct but ```libmmal_vc_client.so``` is not linked.
The file ```libmmal_vc_client.so``` is exist in the path correctly.

So, I added ```LDFLAGS="-Wl,-O1 -Wl,--hash-style=gnu"``` befor autogen as the link https://lists.yoctoproject.org/pipermail/yocto/2014-June/019933.html. Also, I tried to add ```LDFLAGS='-Wl,--no-as-needed'```.
But anyway, output .so file seems same and doesn't work.

thaytan, 2017-06-05:
what's the output from make V=1 ? Is something pulling in -Wl,--as-needed ?

accuware, 2019-10-02:
Did that lead to anything? Having exactly the same problem

accuware, 2019-10-02:
Build and install fine

```
gst-inspect-1.0 | grep rpicamsrc
rpicamsrc:  rpicamsrc: Raspberry Pi Camera Source
rpicamsrc:  rpicamsrcdeviceprovider (GstDeviceProviderFactory)
```

But at runtime:

```
ubuntu@ubiquityrobot:~/gst-rtsp-server-1.8.3/examples$ ./test-launch --gst-debug=3 "( rpicamsrc bitrate=8000000 awb-mode=tungsten preview=false ! video/x-h264, width=640, height=480, framerate=30/1 ! h264parse ! rtph264pay name=pay0 pt=96 )"
stream ready at rtsp://127.0.0.1:8554/test
0:00:29.265774495 22236 0x7310de60 FIXME                default gstutils.c:3766:gst_pad_create_stream_id_internal:<appsrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
0:00:29.270842080 22236 0x75f098c0 ERROR              rpicamsrc RaspiCapture.c:1036:create_camera_component: Failed to create camera component
0:00:29.270904319 22236 0x75f098c0 ERROR              rpicamsrc RaspiCapture.c:1663:raspi_capture_setup: raspi_capture_setup: Failed to create camera component
0:00:29.271168328 22236 0x75f098c0 WARN                 basesrc gstbasesrc.c:3354:gst_base_src_start:<rpicamsrc0> error: Failed to start
0:00:29.271394578 22236 0x75f098c0 WARN                 basesrc gstbasesrc.c:3725:gst_base_src_activate_push:<rpicamsrc0> Failed to start in push mode
0:00:29.271458900 22236 0x75f098c0 WARN                GST_PADS gstpad.c:1106:gst_pad_set_active:<rpicamsrc0:src> Failed to activate pad
0:00:29.274027953 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2552:start_preroll: failed to preroll pipeline
0:00:29.274119827 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2675:start_prepare: failed to preroll pipeline
0:00:29.274646700 22236   0xec6690 WARN               rtspmedia rtsp-media.c:2573:wait_preroll: failed to preroll pipeline
0:00:29.274737533 22236   0xec6690 WARN               rtspmedia rtsp-media.c:2877:gst_rtsp_media_prepare: failed to preroll pipeline
0:00:29.275399823 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2278:default_handle_message: 0x75f20198: got error GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (gstbasesrc.c(3354): gst_base_src_start (): /GstPipeline:media-pipeline/GstBin:bin0/GstRpiCamSrc:rpicamsrc0:
Failed to start)
0:00:29.286751966 22236   0xec6690 ERROR             rtspclient rtsp-client.c:763:find_media: client 0xdc2b48: can't prepare media
0:00:29.288495241 22236   0xec6690 ERROR             rtspclient rtsp-client.c:2288:handle_describe_request: client 0xdc2b48: no media
^C

```

neilyoung, 2019-10-12:
Possible solution here: https://github.com/thaytan/gst-rpicamsrc/issues/92