thaytan/gst-rpicamsrc #52: Cannot compile on Raspberry PI 3

locngy, 2016-07-06:
I got this error when running sudo make install.

pi@raspberrypi:~/p/gst-rpicamsrc-master $ sudo make install
Making install in src
make[1]: Entering directory '/home/pi/p/gst-rpicamsrc-master/src'
make  install-am
make[2]: Entering directory '/home/pi/p/gst-rpicamsrc-master/src'
make[3]: Entering directory '/home/pi/p/gst-rpicamsrc-master/src'
make[3]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/lib/arm-linux-gnueabihf/gstreamer-1.0'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libgstrpicamsrc.la '/usr/lib/arm-linux-gnueabihf/gstreamer-1.0'
libtool: install: /usr/bin/install -c .libs/libgstrpicamsrc.so /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
libtool: install: /usr/bin/install -c .libs/libgstrpicamsrc.lai /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/lib/arm-linux-gnueabihf/gstreamer-1.0
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/lib/arm-linux-gnueabihf/gstreamer-1.0

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to `/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[3]: Leaving directory '/home/pi/p/gst-rpicamsrc-master/src'
make[2]: Leaving directory '/home/pi/p/gst-rpicamsrc-master/src'
make[1]: Leaving directory '/home/pi/p/gst-rpicamsrc-master/src'
Making install in examples
make[1]: Entering directory '/home/pi/p/gst-rpicamsrc-master/examples'
make[2]: Entering directory '/home/pi/p/gst-rpicamsrc-master/examples'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/p/gst-rpicamsrc-master/examples'
make[1]: Leaving directory '/home/pi/p/gst-rpicamsrc-master/examples'
make[1]: Entering directory '/home/pi/p/gst-rpicamsrc-master'
make[2]: Entering directory '/home/pi/p/gst-rpicamsrc-master'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/p/gst-rpicamsrc-master'
make[1]: Leaving directory '/home/pi/p/gst-rpicamsrc-master'

thaytan, 2016-07-06:
There's no error in that output...

locngy, 2016-07-13:
Compilation went fine but I cannot sudo make install.

closed 2016-07-13 by locngy