thaytan/gst-rpicamsrc #57: Bad video output (broken encoding? missing keyframes?)

DavidVentura, 2016-09-20:
Hi. I'm trying to stream with rpicamsrc over the network but I'm getting a lot of errors. This happens over UDP and over TCP so it's not that.

`gst-launch-1.0 rpicamsrc rotation=90 preview=0 $roi ! "video/x-h264, width=$w, height=$h, framerate=24/1, profile=high" ! rtph264pay config-interval=1 ! udpsink host=$1 port=5000`

Short example video: https://gfycat.com/KaleidoscopicThoroughConure

How would I fix this?

thaytan, 2016-09-20:
The default bitrate is the same as raspivid - 17Mbit. Try dropping that a bit (say 1Mbit), and add an h264parse for good measure:

gst-launch-1.0 rpicamsrc rotation=90 preview=0 bitrate=1000000 $roi ! "video/x-h264, width=$w, height=$h, framerate=24/1, profile=high" ! h264parse ! rtph264pay config-interval=1 ! udpsink host=$1 port=5000

tp-m, 2016-09-20:
Maybe also add a queue after the source, so it doesn't block on udpsink syncing against the clock.

DavidVentura, 2016-09-20:
With the queue + h264 parse it works great. Thanks

closed 2016-09-20 by DavidVentura