thaytan/gst-rpicamsrc #58: [Python] toggling preview while running?

DavidVentura, 2016-11-01:
Hi. I made a basic program that plays with some properties but I noticed that I can't change rotation nor preview once the pipeline starts playing. Is this a bug? Intended?

thaytan, 2016-11-02:
The preview is done by building the MMAL pipeline differently, so can't be changed at runtime without adding / removing the preview component. I don't know if MMAL can handle that being done dynamically. The easiest way is probably to leave the preview enabled, but use the preview-x/y/w/h properties to move the preview window offscreen, or maybe shrink it to 0x0 pixels if that works (haven't tried).

Rotation is similar reasoning - rotation 90 degrees changes the width/height of the frame that the encoder is handling which would mean starting a new encode. You should be able to hflip + vflip at runtime though, which lets you get 180 degree rotation.

DavidVentura, 2016-11-02:
Altering preview-x/y/w/h doesn't do anything, either live or before starting the pipeline. Is that supposed to happen?
```
          videosrc.set_property("preview", 1)
          videosrc.set_property("preview-x", 450)
          videosrc.set_property("preview-w", 150)
```
the x/w don't work here

thaytan, 2016-12-01:
Thanks for confirming :)

closed 2016-12-01 by thaytan