thaytan/gst-rpicamsrc #60: Changing bitrate while streaming

Triuman, 2017-01-25:
Hi,

Is there any chance to implement dynamic bitrate as x264enc supports. After setting pipeline to PLAYING state, even I change the bitrate property of rpicamsrc element, it doesn't change it's output bitrate. I'm coding it in C.

Thank you.

thaytan, 2017-01-26:
When I implemented dynamic property changes a while back, I tried the various properties to see which ones the encoder supports changing. Unfortunately, it rejects dynamic bitrate changes. gst-rpicamsrc can't do it without the Rpi firmware supporting it first.

See https://github.com/thaytan/gst-rpicamsrc/blob/master/src/RaspiCapture.c#L1911

6by9, 2017-01-26:
Correct that you can't change the port format whilst the port is enabled, but ```MMAL_PARAMETER_VIDEO_BIT_RATE``` also you to change the bitrate setting at any time.
That's how it is done with the V4L2 driver (https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/media/platform/bcm2835/controls.c#L1103)

```
status = mmal_port_parameter_set_uint32(encoder_output, MMAL_PARAMETER_VIDEO_BIT_RATE, config->bitrate);
```
should do it for you.

Triuman, 2017-01-26:
I added the given line and I can confirm that it's working. This is quite a life saving help. Thank you very much.

@thaytan could you please implement this?

thaytan, 2017-01-27:
Oh cool!

thaytan, 2017-01-27:
Please let me know if you have any trouble with it

commit c74ae97a (2017-01-27): Implement dynamic bitrate update

Use mmal_port_set_parameter_uint32 to update the encoder
bitrate.

Fixes #60
closed 2017-01-27 by thaytan