thaytan/gst-rpicamsrc #62: Drifting video

realies, 2017-06-13:
When using a [B101 board](https://auvidea.com/b101-hdmi-to-csi-2-bridge-15-pin-fpc/) to input 1080p25 or 1080p30 via a GoPro Hero 4 it is noticeable that there is drifting in the captured frames. This is especially apparent when the video stream is muxed with an `alsasrc`. To narrow down the issue I have recorded a 2-minute video only clip ([test.flv.zip](https://github.com/thaytan/gst-rpicamsrc/files/1072082/test.flv.zip)) using GStreamer 1.13.0 (GIT) and  `gst-launch-1.0 -e -vvv rpicamsrc bitrate=1000000 preview=true ! h264parse ! flvmux ! filesink location='test.flv'` on a RPi 3.

The camera is pointed to a smartphone with a 60fps screen playing an [SMPTE time code](http://allensarkisyan.github.io/VideoFrameDocs/) also set to 60fps. What I have observed is that for 30 seconds of recorded video (played back through VLC and gst-play-1.0) only 24 seconds have passed on the recorded screen. Not having access to a standard RPi camera I'm trying to figure out where the issue might be coming from. I have to note that when observing the input via the `preview=true` argument, everything seems normal and as expected.

Can try feeding different resolutions or framerates and also check if there's any difference when `inline-headers` is set to true. Any directions or suggestions are greatly appreciated.

staroselskii, 2017-06-15:
I can second that. It doesn't happen on the camera-number=1 (i.e. Raspberry Pi camera connected to CSI directly). 
This is the command that led me to this reproducing this issue:
`gst-launch-1.0 rpicamsrc camera-number=0 preview=0 bitrate=1000000 ! "video/x-h264, width=1920, height=1080, framerate=24/1, profile=high" ! h264parse ! rtph264pay pt=96 config-interval=5 ! udpsink host=<ip>port=<port>`
Setting different property settings on rtph264pay and changing bitrate haven't yielded any results so far.

Changing `camera-number` to 1 does improve the situation drastically.

We'll be very grateful if you could point us in the direction how this could be fixed. Is this anything you can possibly comment on?

`raspivid -cs 0 -o - | gst ...`
and
`raspivid -cs 1 -o - | gst ...`
work exactly the same. So, unfortunately, I guess it's an issue of the rpicamsrc (which is, by the way, great).

realies, 2017-06-15:
@staroselskii, are you trying `camera-number=1` on a compute module? It would be [strange](https://github.com/thaytan/gst-rpicamsrc/blob/554c9095a2f72405b98317ff72fc706378ff9389/src/gstrpicamsrc.c#L298) if that works on a non compute module setup.

staroselskii, 2017-06-15:
@realies Yes, I test it on a Compute Module v3.

staroselskii, 2017-06-15:
Some update:

1) I attached two cameras (one of them being B101 attached to a laptop via HDMI, though)
2) Launched two instances of the aforementioned gst-launch routine (differing in camera-number and port)
3) Everything seems to be smooth
4) I terminate the instance that is getting input from Raspberry Pi camera
5) The issue gets back!

If I terminate the instance that gets fed from the B101 everything functions as normal.

6) Once I get the gst-launch -cs <rpicamera> back, the lag starts slowly to disappear and the video stream catches up.

Unfortunately, I don't have two camera connectors for Compute module devboard to test a fully symmetrical setup.

I would've certainly started to suspect the B101 module if only the original raspivid didn't work. But it does work as expected. Any clues? Something's going on with the clock? I'm no expert here at all. But I tried my best to supply as much as I could to share with you!

staroselskii, 2017-06-29:
@realies 

Hello there! Is there any chance you'll look into the issue? I'd be happy to debug but I just don't know where to start. I'd be very grateful for any hints.

thaytan, 2017-06-30:
The main difference between raspivid and rpicamsrc timestamping is that rpicamsrc tries to use the hardware timestamp and MMAL_PARAMETER_SYSTEM_TIME reported by the rpi hardware to timestamp output buffers (which should reduce jitter in the output frame timings). If those timestamp lie, it'd produce an effect like this.

To change that, rpicamsrc needs modifying.

In src/gstrpicamsrc.c line 525, change gst_base_src_set_do_timestamp (..., FALSE) to TRUE
In src/RaspiCapture.c comment out line 972 setting GST_BUFFER_PTS ()

Some log output of rpicamsrc (without those changes) would also be handy. Run gst-launch-1.0 with GST_DEBUG set:

GST_DEBUG=rpicamsrc:6 gst-launch-1.0 rpicamsrc etc.....

thaytan, 2017-06-30:
Try updating to the newest commit 4fc608 - I added a 'use-stc' property you can set to false to get the default timestamping by the GStreamer clock. See if it helps.

realies, 2017-07-02:
@thaytan, thanks for the follow up. I can confirm that using a stopwatch, the recording of a SMPTE time code stays visually in sync.

thaytan, 2017-07-02:
The recording stays in sync under which conditions?

realies, 2017-07-03:
@thaytan, redoing the recordings with #4fc608 reveals that having both `use-stc=true` and `use-stc=false` make it visually in sync with the latter introducing some subtle glitches but otherwise on time.

`timeout 130 gst-launch-1.0 -e -vvv rpicamsrc bitrate=1000000 use-stc=false ! video/x-h264,width=1920,height=1080,framerate=25/1,profile=high ! h264parse ! flvmux ! filesink location='4fc608_stc-false.flv'`
[4fc608_stc-false.flv.zip](https://github.com/thaytan/gst-rpicamsrc/files/1118218/4fc608_stc-false.flv.zip)

`timeout 130 gst-launch-1.0 -e -vvv rpicamsrc bitrate=1000000 use-stc=true ! video/x-h264,width=1920,height=1080,framerate=25/1,profile=high ! h264parse ! flvmux ! filesink location='4fc608_stc-true.flv'`
[4fc608_stc-true.flv.zip](https://github.com/thaytan/gst-rpicamsrc/files/1118219/4fc608_stc-true.flv.zip)

`timeout 130 gst-launch-1.0 -e -vvv rpicamsrc bitrate=1000000 ! video/x-h264,width=1920,height=1080,framerate=25/1,profile=high ! h264parse ! flvmux ! filesink location='554c909.flv'`
[554c909.flv.zip](https://github.com/thaytan/gst-rpicamsrc/files/1118217/554c909.flv.zip)

staroselskii, 2017-07-03:
I can also also confirm that `use-stc=false` fixes the issue for me. Very thankful for the time you've spent fixing this!

realies, 2017-07-03:
@staroselskii, #4fc608 should work fine/better without `use-stc=false`.

staroselskii, 2017-07-03:
@realies 
As a matter of fact, I did test with and without this setting set and got better results that I'd got before (sorry for the confusion). You happened to update the thread faster that I got back to the keyboard to write back. It all boils down to the fact that this commit fixes the issue for me. I reverted the commit, tested and the drift got back.

closed 2017-07-03 by realies

thaytan, 2017-07-04:
Thanks for confirming. For my own future reference - the problem was rpicamsrc applying (bogus) duration and timestamps to encoder overflow packets - any output from the encoder larger than 64KB gets split into a separate buffer.