thaytan/gst-rpicamsrc #64: splitter_port

StickyDigit, 2017-07-23:
I've grepped around in the code, and dug around the net for clues but don't see any reference to splitter_port as used in picamera.
Is there any plan to make this available in gst-rpicamsrc?
I'm hoping to pull multiple streams from the camera at different resolutions with different ROIs.

realies, 2017-07-24:
This would be very useful if possible by the hardware.

StickyDigit, 2017-07-24:
According to the picamera docs it is. IIRC I played with it a couple of years ago. Picamera has four splitter ports, allowing to grab a full-res still on port 0 while having video preview on say port 1, and two further streams of video. Looks like this is used in gst-rpicamsrc but I can't quite fathom how, and it doesn't appear to be exposed.

thaytan, 2017-08-02:
The rpicamsrc source is basically a meld of raspivid and raspistill mashed together, and has the same feature set. I don't have any plans to expose multiple output streams, but will take clean patches for it. The GStreamer side might get tricky - the underlying basesrc class really only expects 1 output stream, so it could get hairy.

StickyDigit, 2017-08-06:
Ok. Thanks.
I may attempt a fork if I can digest the gstreamer end sufficiently. Perhaps something along the lines of a "slave mode", where a second rpicamsrc element with arg slave=3 would use stream 3 from the master instance, and disallow any settings (i.e. exposure) which make no sense on a slave.

thaytan, 2017-08-09:
That approach might work. The other thing to consider is how quickly you'll run out of GPU resources. As far as I understand, the GPU can only manage a maximum of encoding a single stream at 1080p @ 30 fps. It can do multiple smaller streams, or lower framerates - but you're dividing the GPU core capability between encoder instances. I'm also not sure how much of the ROI selection is done by setting cropping properties on the camera sensor, and how much you can affect it before handing off frames to the encoder.

thaytan, 2017-10-10:
Please re-open if you need any further help or end up making a patch.

closed 2017-10-10 by thaytan