thaytan/gst-rpicamsrc #65: Handle unexpected camera control callback event

staroselskii, 2017-07-24:
Hello there!

Is there a way to propagate this kind of error further to gstreamer?

```
0:00:48.536833422  8002   0xc79960 ERROR              rpicamsrc RaspiCapture.c:757:camera_control_callback: Received unexpected camera control callback event, 0x4f525245
```

It happens for example when camera is turned off while streaming.

I also noticed that when camera is turned back on, bit stream doesn't get reinitialized and hence the data form the camera doesn't get sent.

thaytan, 2017-08-03:
How do you trigger that? You have a camera with an external power switch?

staroselskii, 2017-08-03:
Yep, that is exactly the situation. I guess the same thing would pop up if Raspberry Pi Camera is pulled out.

thaytan, 2017-08-06:
The GStreamer way to propagate that will be to generate a GST_MESSAGE_ERROR on the bus so the application can find out and restart the pipeline.

thaytan, 2017-11-10:
I might do a bit of work later to propagate a correct and useful error on the bus rather than the generic one, but this looks like it works fine for your use case already, so I think that's a nice-to-have. Thanks!

closed 2017-11-10 by thaytan