thaytan/gst-rpicamsrc #66: SIGSEGV asking for h264 stream vrom rpicamsrc

ndepal, 2017-07-28:
I just installed `gst-rpicamsrc` on a Pi Z (no wifi) running a fresh Raspbian Jessie install.

The following pipeline works fine:
```
gst-launch-1.0 -v rpicamsrc ! omxh264enc ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.0.25 port=5000 async=false
```

However, I want to leave out `omxh264enc` since `rpicamsrc` is capable of providing an h264 stream directly, thus changing the pipeline to the following (simply leaving `omxh264enc`):

```
gst-launch-1.0 -v rpicamsrc !  rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.0.25 port=5000 async=false
```

This segfaults with the following output:

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_int: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_int: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_fraction: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_int: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_int: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:8168): GStreamer-CRITICAL **: gst_structure_fixate_field_nearest_fraction: assertion 'gst_structure_has_field (structure, field_name)' failed
/GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au"
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x1129f20 (EINVAL)
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.044498611
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Caught SIGSEGV
#0  0xb6cf9ef8 in __libc_waitpid (pid=8179, stat_loc=0xbec86a78, options=0)
#1  0xb6d2cec8 in g_on_error_stack_trace ()
#2  0x00014c08 in fault_spin () at gst-launch.c:99
#3  fault_handler_sighandler (signum=<optimized out>) at gst-launch.c:90
#4  <signal handler called>
#5  __GI___pthread_mutex_lock (mutex=0x13c) at pthread_mutex_lock.c:66
#6  0xb67e7d14 in mmal_component_disable () from /opt/vc/lib/libmmal_core.so
Spinning.  Please run 'gdb gst-launch-1.0 8168' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

Any advice?

thaytan, 2017-08-01:
You can work around that by putting a video filter in the path to tell rpicamsrc what to output. You also need an h264parse before the rtp payloader:

rpicamsrc ! video/x-h264,width=1920,height=1080,framerate=30/1 ! h264parse ! rtph264pay ...

If I get a chance, I'll take a look at fixing the crash.

ndepal, 2017-08-02:
I've noticed that specifying caps works around the issue.
Can you shed some light on when one does or does not need an `h264parse` element?

thaytan, 2017-08-04:
Actually, it will work without in this case, because rpicamsrc outputs NAL aligned packets that are already acceptable for RTP payloading. It doesn't hurt to include though - h264parse will do passthrough with almost no overhead where it can.

divinebovine, 2017-08-08:
I'm having this problem as well, running on a pi zero w
```
$ gst-launch-1.0 -v rpicamsrc ! video/x-h264,width=1920,height=1080,framerate=30/1 ! h264parse ! rtph264pay config-interval=1 pt=96 ! gdppay ! udpsink host=192.168.1.9 port=9999
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0.GstPad:src: caps = "video/x-h264\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ framerate\=\(fraction\)30/1\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)nal\,\ profile\=\(string\)baseline"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ framerate\=\(fraction\)30/1\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)nal\,\ profile\=\(string\)baseline"
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = "video/x-h264\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ framerate\=\(fraction\)30/1\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)nal\,\ profile\=\(string\)baseline"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ framerate\=\(fraction\)30/1\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)nal\,\ profile\=\(string\)baseline"
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.158667974
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

thaytan, 2017-08-08:
@divinebovine That looks completely unrelated to this bug, and is probably just that you didn't allocated enough GPU memory in /boot/config.txt

thaytan, 2017-10-10:
Closing as no further information provided

closed 2017-10-10 by thaytan