thaytan/gst-rpicamsrc #69: Internal Data Stream Error

crossan007, 2017-11-14:
This is with a Raspi3, and Camera module v2.1

Pipeline: ``` gst-launch-1.0 rpicamsrc bitrate=10000 sensor-mode=1 preview=false preview-encoded=false use-stc=false do-timestamp=true ! fakesink      ```
result: 

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.330214825
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

with GST_DEBUG=4, the relevant failure output is as follows:
```
0:00:00.372352083   837 0x72a02920 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<pipeline0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.472158505   837  0x1af9980 INFO                 basesrc gstbasesrc.c:2743:gst_base_src_loop:<rpicamsrc0> pausing after gst_base_src_get_range() = error
0:00:00.472223505   837  0x1af9980 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc0> error: Internal data stream error.
0:00:00.472253766   837  0x1af9980 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc0> error: streaming stopped, reason error (-5)
0:00:00.472343765   837  0x1af9980 INFO        GST_ERROR_SYSTEM gstelement.c:1890:gst_element_message_full_with_details:<rpicamsrc0> posting message: Internal data stream error.
0:00:00.472451994   837  0x1af9980 INFO        GST_ERROR_SYSTEM gstelement.c:1917:gst_element_message_full_with_details:<rpicamsrc0> posted error message: Internal data stream error.
0:00:00.472543036   837  0x1af9980 INFO                    task gsttask.c:318:gst_task_func:<rpicamsrc0:src> Task going to paused
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.329523132
```

```gst-inspect-1.0 rpicamsrc``` shows the expected output:
```
gst-inspect-1.0 rpicamsrc
Factory Details:
  Rank                     none (0)
  Long-name                Raspberry Pi Camera Source
  Klass                    Source/Video
  Description              Raspberry Pi camera module source
  Author                   Jan Schmidt <jan@centricular.com>

Plugin Details:
  Name                     rpicamsrc
  Description              Raspberry Pi Camera Source
  Filename                 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
  Version                  1.0.0
  License                  LGPL
  Source module            gstrpicamsrc
  Binary package           GStreamer
  Origin URL               http://gstreamer.net/

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstBaseSrc
                         +----GstPushSrc
                               +----GstRpiCamSrc
```

crossan007, 2017-11-14:
Execution seems to end after ~ 0:00:00.336200142 +/- 00:00.0001

crossan007, 2017-11-14:
4fc608 seems to get stuck at ```0:00:00.371406592  3558 0x72a02920 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<pipeline0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)``` when starting the pipeline.

crossan007, 2017-11-14:
currently trying with a "fresh" version of GStreamer from git with steps from @thaytan:

step 1 is 'apt-get build-dep' for all the modules, to get the build dependencies installed
step 2 is use https://cgit.freedesktop.org/gstreamer/gstreamer/plain/scripts/create-uninstalled-setup.sh to compile the GStreamer modules on top

crossan007, 2017-11-14:
This seems to work when using gst-launch-1.0 version 1.13.0 instead of gst-launch-1.0 version 1.10.4

thaytan, 2017-11-15:
I don't understand what difference 1.13.0 would make vs 1.10.4. As I mentioned on IRC, I've tried rpicamsrc with lots of versions of GStreamer and not encountered that problem. It definitely works for me against 1.4.4 and git master 1.13.0.1

crossan007, 2017-11-15:
super weird.   I'll try it with a fresh raspian image tonight.

crossan007, 2017-11-17:
Not exactly sure what caused my first attempt to fail so miserably, but a clean image of ```2017-09-07-raspbian-stretch-lite.zip``` with the install guide in this repo's README.md worked with only one small hiccup:

The first time I ran it, I got one Internal data stream error:
```
pi@raspberrypi:~ $ gst-launch-1.0 rpicamsrc bitrate=1000000 ! filesink location=test.h264
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
streaming stopped, reason error (-5)
Execution ended after 0:00:03.398609582
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

All subsequent runs seem to be fine.

I'll close the issue.

closed 2017-11-17 by crossan007

thaytan, 2017-11-17:
I think the generic internal data error is the camera taking too long to start and the new timeout code triggering an error. Perhaps 500ms is still not long enough of a timeout for the first frame. In any case, I need to make it post a more useful and recognisable error when the timeout happens.