thaytan/gst-rpicamsrc #72: gst-launch-1.0: command not found

HibikiKai, 2018-02-10:
pi@raspberrypi:~/gst-rpicamsrc $ sudo make install
Making install in src
make[1]: Entering directory '/home/pi/gst-rpicamsrc/src'
make  install-am
make[2]: Entering directory '/home/pi/gst-rpicamsrc/src'
make[3]: Entering directory '/home/pi/gst-rpicamsrc/src'
make[3]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/lib/arm-linux-gnueabihf/gstreamer-1.0'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libgstrpicamsrc.la '/usr/lib/arm-linux-gnueabihf/gstreamer-1.0'
libtool: install: /usr/bin/install -c .libs/libgstrpicamsrc.so /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
libtool: install: /usr/bin/install -c .libs/libgstrpicamsrc.lai /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/lib/arm-linux-gnueabihf/gstreamer-1.0
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/lib/arm-linux-gnueabihf/gstreamer-1.0

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[3]: Leaving directory '/home/pi/gst-rpicamsrc/src'
make[2]: Leaving directory '/home/pi/gst-rpicamsrc/src'
make[1]: Leaving directory '/home/pi/gst-rpicamsrc/src'
Making install in examples
make[1]: Entering directory '/home/pi/gst-rpicamsrc/examples'
make[2]: Entering directory '/home/pi/gst-rpicamsrc/examples'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/gst-rpicamsrc/examples'
make[1]: Leaving directory '/home/pi/gst-rpicamsrc/examples'
make[1]: Entering directory '/home/pi/gst-rpicamsrc'
make[2]: Entering directory '/home/pi/gst-rpicamsrc'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/gst-rpicamsrc'
make[1]: Leaving directory '/home/pi/gst-rpicamsrc'
pi@raspberrypi:~/gst-rpicamsrc $ sudo ldconfig 
pi@raspberrypi:~/gst-rpicamsrc $ sudo ldconfig 
pi@raspberrypi:~/gst-rpicamsrc $ sudo ldconfig 
pi@raspberrypi:~/gst-rpicamsrc $ sudo ldconfig 
^[[A
pi@raspberrypi:~/gst-rpicamsrc $ sudo ldconfig 
pi@raspberrypi:~/gst-rpicamsrc $ gst-launch-1.0 rpicamsrc bitrate=1000000 fullscreen=0 ! video/x-h264,width=640,height=480,framerate=25/1 ! filesink location=test.h264
bash: gst-launch-1.0: command not found

tp-m, 2018-02-10:
`gst-launch-1.0` is provided by the `gstreamer1.0-tools` package on raspbian.

endlessbest, 2018-07-12:
just do this: 
sudo apt-get install gstreamer1.0-tools

PanderMusubi, 2021-01-22:
Can this issue be closed?

thaytan, 2021-01-24:
Yes, thanks.

closed 2021-01-24 by thaytan