thaytan/gst-rpicamsrc #74: mmal: mmal_vc_component_enable: failed to enable component: ENOSPC

dherman516, 2018-04-04:
I get this when restarting a RTSP steam (using  https://github.com/GStreamer/gst-rtsp-server.git).
How can I enable logging to root cause the failure?

Hexagenic, 2018-04-05:
You can set the GST_DEBUG environment variable. For example
```
GST_DEBUG=3 gst-launch rpicamsrc ...
```

dherman516, 2018-04-05:
0:00:50.048022819 27716 0x72d18180 FIXME                default gstutils.c:3826:gst_pad_create_stream_id_internal:<appsrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:50.065631169 27716 0x72d180c0 FIXME                default gstutils.c:3826:gst_pad_create_stream_id_internal:<rpicamsrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:50.371974630 27716  0x102ce90 FIXME              rtspmedia rtsp-media.c:3581:gst_rtsp_media_suspend: suspend for dynamic pipelines needs fixing
0:00:50.401000094 27716 0x75114920 FIXME                default gstutils.c:3826:gst_pad_create_stream_id_internal:<appsrc1:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:50.414884389 27716 0x75114860 FIXME                default gstutils.c:3826:gst_pad_create_stream_id_internal:<rpicamsrc1:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
0:00:50.428601758 27716 0x75114860 ERROR              rpicamsrc RaspiCapture.c:1257:raspi_capture_set_format_and_start: camera component couldn't be enabled
0:00:50.430689410 27716 0x75114860 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc1> error: Internal data stream error.
0:00:50.431852221 27716 0x75114860 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc1> error: streaming stopped, reason error (-5)
0:00:50.433348572 27716 0x75c04e30 WARN               rtspmedia rtsp-media.c:2433:default_handle_message: 0x75c2ca60: got error Internal data stream error. (gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:media-pipeline/GstBin:bin1/GstRpiCamSrc:rpicamsrc1:
streaming stopped, reason error (-5))

dherman516, 2018-04-05:
sudo vcdbg log msg

>1209749.541: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 32, power enable gpio = 41
1209750.259: camsubs: Camera not found
1209750.297: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 32, power enable gpio = 41
1209751.931: camsubs: Camera found OK
1209754.044: gpioman: gpioman_get_pin_num: pin CAMERA_LED not defined
1209754.471: brfs: File read: 1612 bytes
1209965.644: mmal: mmal_vll_load: could not load VLL 'videnc.vll':
1210455.709: gpioman: gpioman_get_pin_num: pin CAMERA_LED not defined
1210456.237: mmalsrv: send_buffer_to_host: tx failed:size 292 st -1

dherman516, 2018-04-05:
Same exact error on two different pi boards

thaytan, 2018-04-05:
I think this might mean that the previous client hasn't shut down yet - so there's a 2nd rpicamsrc trying to use the camera when it's already in use. When I share the camera via gst-rtsp-server, I set gst_rtsp_media_factory_set_shared(TRUE) so only 1 instance is shared to all clients.

dherman516, 2018-04-05:
how do you do that?

> I set gst_rtsp_media_factory_set_shared(TRUE)

hanifizzudinrahman, 2021-09-12:
Hello brother, finally i can solve this problem after a lot of hours troubleshooting.

I already share it in my YouTube ([https://youtu.be/on8UlkRvLok](https://youtu.be/on8UlkRvLok))

This problem only occur in camera raspberry pi V2.1 (IMX219) from **Element14**. But in camera raspberry pi V2.1 from **OKdo**, i don't get this error

So, the solution when error
"mmal: mmal_vc_component_enable: failed to enable component: **ENOSPC**
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates"

You need some initialization with a works camera (me camera from **OKdo**).
So, the step is
1. Plug your works camera to Raspberry Pi (me using Raspberry Pi 4B)
2. Turn on your Raspberry Pi
3. check `vcgencmd get_camera `to make sure your hardware is detected the camera
4. capture image using `raspistill -v` and camera will capture the image
5. Then replace the works camera with ENOSPC Error Camera (from **Element14**)
6. then try to capture the image `raspistill -v` and the ENOSPC Error will not occur

**Notes:**
- But, after the ENOSPC Error Camera works, and then you restart/shutdown the raspberry pi, the error will occur again :(

**OS Raspberry Pi 4 Model B (4GB)**
- Linux raspberrypi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux

Z1scko, 2023-07-20:
Hello,
So I exactly had this problem with my Pi4 2Gb and the V3 official camera. Unable to find a solution. I then tried with another camer: the High Quality camera (official product too), and everything worked fine.
It might be a factory issue.