thaytan/gst-rpicamsrc #76: Internal Data Stream Error (-5)

xloem, 2018-06-09:
Every time I run my pipeline, it goes for a while, and then eventually crashes with, "streaming stopped, reason error (-5)".  I've seen this error mentioned in closed issues, but the general consensus seems to be that this is a startup timeout issue, which it is not for me.  Streaming runs for many minutes before eventually crashing.

I ran this pipeline:
``` gst-launch-1.0 --quiet rpicamsrc preview=false annotation-mode=$((0x40c)) annotation-text-size=16 ! video/x-h264,width=1296,height=972,framerate=42/1 ! h264parse ! matroskamux ! fdsink ```

It ran fine for 6.5 minutes, then suddenly gave this:
```
0:06:30.089596969 30998  0x17ea6c0 INFO                 basesrc gstbasesrc.c:2743:gst_base_src_loop:<rpicamsrc0> pausing after gst_base_src_get_range() = error              
0:06:30.089877227 30998  0x17ea6c0 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc0> error: Internal data stream error.                       
0:06:30.089923425 30998  0x17ea6c0 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<rpicamsrc0> error: streaming stopped, reason error (-5)
0:06:30.090105038 30998  0x17ea6c0 INFO        GST_ERROR_SYSTEM gstelement.c:1890:gst_element_message_full_with_details:<rpicamsrc0> posting message: Internal data stream error.                
0:06:30.090483732 30998  0x17ea6c0 INFO        GST_ERROR_SYSTEM gstelement.c:1917:gst_element_message_full_with_details:<rpicamsrc0> posted error message: Internal data stream error.
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: Internal data stream error.                                                        
Additional debug info:                                                                                                                                                                          
gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:                                                                        
streaming stopped, reason error (-5)                                                                                                      
0:06:30.091606119 30998  0x17ea6c0 INFO                    task gsttask.c:318:gst_task_func:<rpicamsrc0:src> Task going to paused                                
0:06:30.091633410 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<fdsink0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:06:30.091843304 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<fdsink0> completed state change to PAUSED
0:06:30.091911429 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<fdsink0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.092028980 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'fdsink0' changed state to 3(PAUSED) successfully
0:06:30.092114969 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<matroskamux0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:06:30.092186791 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<matroskamux0> completed state change to PAUSED
0:06:30.092241530 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<matroskamux0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.092318717 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'matroskamux0' changed state to 3(PAUSED) successfully
0:06:30.092410800 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<h264parse0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:06:30.092477674 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<h264parse0> completed state change to PAUSED
0:06:30.092551371 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<h264parse0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.092625850 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'h264parse0' changed state to 3(PAUSED) successfully
0:06:30.092696005 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<capsfilter0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:06:30.092801630 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<capsfilter0> completed state change to PAUSED
0:06:30.092854546 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<capsfilter0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.092933295 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'capsfilter0' changed state to 3(PAUSED) successfully
0:06:30.093035690 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rpicamsrc0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:06:30.093117825 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<rpicamsrc0> completed state change to PAUSED
0:06:30.093174335 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rpicamsrc0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.093253657 30998  0x170fe00 INFO              GST_STATES gstbin.c:3012:gst_bin_change_state_func:<pipeline0> child 'rpicamsrc0' changed state to 3(PAUSED) successfully without preroll
0:06:30.093340219 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<pipeline0> completed state change to PAUSED
0:06:30.093395948 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<pipeline0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:06:30.093674174 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<fdsink0> current PAUSED pending VOID_PENDING, desired next READY
0:06:30.094145472 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<fdsink0> completed state change to READY
0:06:30.094240784 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<fdsink0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:06:30.094354846 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'fdsink0' changed state to 2(READY) successfully
0:06:30.094475522 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<matroskamux0> current PAUSED pending VOID_PENDING, desired next READY
0:06:30.094698332 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<matroskamux0> completed state change to READY
0:06:30.094758905 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<matroskamux0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:06:30.094838071 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'matroskamux0' changed state to 2(READY) successfully
0:06:30.094917445 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<h264parse0> current PAUSED pending VOID_PENDING, desired next READY
0:06:30.095369473 30998  0x170fe00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<h264parse0> completed state change to READY
0:06:30.095441191 30998  0x170fe00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<h264parse0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:06:30.095556086 30998  0x170fe00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'h264parse0' changed state to 2(READY) successfully
0:06:30.095709314 30998  0x170fe00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<capsfilter0> current PAUSED pending VOID_PENDING, desired next READY
[... cut ...]
0:06:33.907775066 30998  0x170fe00 INFO         GST_REFCOUNTING gstelement.c:3077:gst_element_finalize:<pipeline0> 0x1804068 finalize parent                                 
0:06:33.907873658 30998  0x170fe00 INFO                GST_INIT gst.c:989:gst_deinit: deinitializing GStreamer                                          
0:06:33.920943497 30998  0x170fe00 INFO                GST_INIT gst.c:1130:gst_deinit: deinitialized GStreamer                                                                                    
```

Swap-File, 2018-06-10:
You might want to try adding a queue to the pipeline and seeing if anything changes.  I've noticed that pipelines without a queue can be iffy, adding a queue will decouple the input and the output.

xloem, 2018-06-11:
I tried just throwing 'queue' in different spots with no options, but it didn't help.  I tried running under gdb and the error occurs much more rapidly; it seems to do with latency.  I ended up fixing it by changing the codec to x-raw and adding omxh264enc, but the cpu usage was too high for me, so I've switched back to plain raspicamvid for now.

Swap-File, 2018-06-11:
It could have something to do with whatever is consuming the data from the fdsink.  Where is the data going from there?  

Alternatively, if you think the problem is with rpicamsrc you could try piping raspivid into gstreamer, but the CPU usage is slightly higher than normal.

Something like this (going from memory):

/usr/bin/raspivid -w 1296 -h 972 -i record -fps 42 -t 0 -o - | /usr/bin/gst-launch-1.0 fdsrc ! queue ! h264parse ! matroskamux ! fdsink

thaytan, 2018-06-11:
Take a look at commit 4e5729 and perhaps try expanding that timeout further from 500ms

shortstheory, 2018-06-12:
I had the exact same issue. I fixed it by setting the timeout as 1000ms and now it works swimmingly.