thaytan/gst-rpicamsrc #80: Camera capture timed out

quantum8, 2018-08-13:
I'm trying to enable orientation on the pipeline. 

My stream seems to work when using:
`gst-launch-1.0 -v rpicamsrc name=src preview=0 exposure-mode=night fullscreen=0 bitrate=1000000 annotation-mode=time+date annotation-text-size=20 ! video/x-h264,width=960,height=540,framerate=8/1 ! queue max-size-bytes=0 max-size-buffers=0 ! h264parse ! rtph264pay config-interval=1 pt=96 ! queue ! udpsink host=127.0.0.1 port=5004  sync=false`

But inserting the rotation parameter I get an error:
`gst-launch-1.0 -v rpicamsrc name=src preview=0 exposure-mode=night fullscreen=0 bitrate=1000000 rotation=180 annotation-mode=time+date annotation-text-size=20 ! video/x-h264,width=960,height=540,framerate=8/1 ! queue max-size-bytes=0 max-size-buffers=0 ! h264parse ! rtph264pay config-interval=1 pt=96 ! queue ! udpsink host=127.0.0.1 port=5004  sync=false`

`Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstRpiCamSrc:src.GstPad:src: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:src: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-h264, width=(int)960, height=(int)540, framerate=(fraction)12/1, stream-format=(string)byte-stream, alignment=(string)nal, profile=(string)constrained-baseline
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:src: Camera capture timed out.
Additional debug info:
gstrpicamsrc.c(1445): gst_rpi_cam_src_create (): /GstPipeline:pipeline0/GstRpiCamSrc:src:
Waiting for a buffer from the camera took too long.
Execution ended after 0:00:00.573751795
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...`

Is there something that I need to add or change here?

aeozyalcin, 2019-01-17:
I am running into exactly the same issue as you, but I am running Raspbian Jessie... Were you able to solve the issue?

aeozyalcin, 2019-01-17:
Ok I was able to solve the problem by following: https://github.com/thaytan/gst-rpicamsrc/issues/77#issuecomment-403443858

There is another issue on here for a similar problem, and increasing timeout to 1500 worked for me. Good luck!

Abu-Abdullah, 2019-05-21:
I'm facing the same issue with normal pipeline:

```
gst-launch-1.0 -v rpicamsrc ! video/x-h264,width=640,height=480,framerate=8/1,profile=high ! h264parse ! splitmuxsink location=/home/pi/snapshots_tmp/test.mp4 max-size-time=30000000000 max-size-bytes=2000000
```

this is on Pi0
chaning the timeout to 1500 solved the issue. i believe this should be the default number not 500

jeremyfritzen, 2019-11-13:
Hi!
I have the same issue but I don't know how to change the timeout to 1500. Could you explain please? Thanks!

thaytan, 2019-11-14:
You need to edit this line:
https://github.com/thaytan/gst-rpicamsrc/blob/master/src/RaspiCapture.c#L941
and change the timeout from 500ms to 1500ms

thaytan, 2019-11-18:
This camera timeout needs a more comprehensive fix when I get time. I'm not sure if it's possible to make a solution that auto-adjusts the timeout, but I could make it a property with a sufficiently large default, and let people that need it smaller make it smaller.

WisdomPill, 2020-07-11:
Any update on this?

I am facing the same issue with opencv, this is my pipeline

```
rpicamsrc preview=false ! video/x-raw,format=BGR,width=640,height=480 ! queue name=sink leaky=2 ! appsink max-buffers=5 drop=true sync=false
```

```
[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module rpicamsrc0 reported: Camera capture timed out.
[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

WisdomPill, 2020-07-13:
Rasing the timeout to 1500 fixed the issue

thaytan, 2020-07-13:
I would like to make it a property, with a higher default

WisdomPill, 2020-07-22:
I correct myself, 1500 ms of timeout just delays the issue.
With raspberry 3b+ happens more often (every 5 minutes) with the standard 500 while with the increased value happens more sporadically.
While with the raspberry 4 it does not happen even with the standard value of 500.

I will try implementing what you have suggested here https://github.com/thaytan/gst-rpicamsrc/issues/77#issuecomment-405162250

thaytan, 2020-07-22:
what are your lighting conditions like? 1.5seconds before a frame arrives is a very long time!

WisdomPill, 2020-07-23:
They are pretty good, although the application runs 24/7 outdoors so it is possible to have a pitch dark environment.
It does not block only when there is no light, also when there is.

thaytan, 2020-07-24:
That's pretty interesting - I do wonder what's going on when the timeout happens. 1.5 seconds is a really long time to get stuck and not get even a single packet out!

WisdomPill, 2020-08-09:
It is more than two weeks that it is working flawlessly, I added the change and I am proposing a PR, should I do that in the new version of gstreamer as well?

thaytan, 2020-08-10:
Simply removing the timeout isn't a fix - it's what was there before the timeout was added in the first place.

WisdomPill, 2020-08-10:
Why it was added? Are there some other tests that I can do?

thaytan, 2020-08-10:
I think the original request was for a CSI HDMI-input adapter board and they wanted to detect an error if the HDMI signal went away. The other use case the timeout captures is if the firmware locks up - without a timeout it's possible to get stuck forever trying to dequeue.

I think the best option is to connect the timeout to a property on the element so it can be adjusted as needed, and give it a suitably large default.

I still would like to know why you'd ever get 1.5 seconds gap between frames though. That's far too much - way more than a frame duration at any sensible framerate.