thaytan/gst-rpicamsrc #81: Refuses to capture at 40 fps in mode 4, works in raspivid

xloem, 2018-08-18:
I had a v2 camera (unfortunately broken as of today), and was trying to capture at 1640x1232 at 40fps (mode 4).

Unfortunately, rpicamsrc crashes with these settings with an mmal error EINVAL I believe.  However, the settings work fine with raspivid.

thaytan, 2018-08-20:
What are your command lines?

xloem, 2018-08-30:
hey, I haven't figured out my busted camera to check yet, but from memory I was manually setting the width, height, and framerate, but not setting the camera mode.  Checking the source it seemed like gst-rpicamsrc may not recognize v2 mode configurations.

My next attempt, when I make enough time to resolve the broken camera issue, will be to try manually setting the mode number.

thaytan, 2018-09-02:
I don't think there's anywhere that the rpicamsrc code needs to know individual camera sensor modes - it's just an integer that gets passed through to MMAL, or left as 0 for the driver to auto-select