thaytan/gst-rpicamsrc #84: Issue in building webrtc-unidirectional example

jagadishkamathk, 2019-01-20:
I'm trying to build the webrtc-unidirectional-h264.c and I see it isn;t enabled by default. Configure excludes it with checking for GST_WEBRTC_EXAMPLE... no
WHat is the way to enable this in autogen or configure?

thaytan, 2019-01-22:
and a new enough GStreamer 1.14 with WebRTC support

jagadishkamathk, 2019-02-07:
Thank you. That seems to work. Just curious. How does the GST_WEBRTC_EXAMPLE environment variable gets set when these two packages get installed?

Also, webrtc exchange seems to happen but no media is seen on either chrome or firefox. That could be an entirely different issue though

kevkid, 2019-02-12:
@jagadishkamathk I am seeing the same issue, Cant seem to get media to display. Maybe chrome, or firefox changes something? I have had much difficulty getting h264 to render in the webpage.

soft-machine, 2019-02-13:
Hello. I struggled for a good 5 hours trying to deal with the same issue. I finally got the thing built. I still need to try the examples (get my head around the RTC stuff in general). In the meantime I am posting here the documentation of my own process to get things working. We'll see if it holds together in the coming days:

Setup
======
1) dcfldd latest raspbian stretch lite image to SD card
2) change references in apt repositories to buster, replacing stretch (/etc/apt/sources.list & /etc/apt/sources.list.d/raspi.list)
3) apt-get update && apt-get upgrade && apt-get dist-upgrade

Then prepare dependencies, etc.
===========================
Not sure if all the following packages are necessary, but the combination worked for me to get Thaytan's gst-rpicamsrc compiled and working with full examples (specifically GST_WEBRTC_EXAMPLE)

**sudo apt-get install -y gstreamer1.0-x libgstreamer-plugins-base1.0-dev gstreamer1.0-tools gstreamer1.0-nice gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-plugins-good libgstreamer1.0-dev git libglib2.0-dev libgstreamer-plugins-bad1.0-dev libsoup2.4-dev libjson-glib-dev libnice-dev autoconf libtool gtk-doc-tools libssl-dev libreadline-dev libsrtp0-dev**

Also manually installed openwebrtc-gst-plugins following guidelines listed here:

[Build-GstWebRTC](https://developer.ridgerun.com/wiki/index.php?title=GstWebRTC_-_Building_GstWebRTC )

Not sure if this was necessary however. I fudged in the instruction process, cloning and installing GStreamer SCTP elements without first building the SCTP libraries as advised here (which makes me think it may be unnecessary)

After these steps I could then finally get the autogen.sh to respond GST_WEBRTC_EXAMPLE... yes

The following link provided me additional help for finding required dependencies:
[centricular/gstwebrtc-demos](https://github.com/centricular/gstwebrtc-demos)

jagadishkamathk, 2019-02-13:
But where you able to see the Raspi camera output on your browser?

soft-machine, 2019-02-13:
No. Not yet. Executing the script doesn't seem to even turn on the camera (no indicator light response nor video feedback on HDMI screen as is usually the case when doing gst-launch udp/tcp tests).

thaytan, 2019-02-19:
The camera doesn't turn on until you connect with a client and the streaming starts. It's a on my wish-list to extend the example with capabilities like record to disk and multiple client support (atm the 2nd client will fail because it tries to open the camera a 2nd time)

kevkid, 2019-06-17:
I get this issue every time I try to compile the webrtc-unidirectional example.

```
make[2]: Entering directory '/home/pi/gst-rpicamsrc/examples'
  CCLD     webrtc-unidirectional-h264
libtool:   error: cannot find the library '/opt/gstreamer/lib/lib-pthread.la' or unhandled argument '/opt/gstreamer/lib/lib-pthread.la'
Makefile:397: recipe for target 'webrtc-unidirectional-h264' failed
```
I have no idea what to do next with this?

thaytan, 2019-06-19:
I don't know why you'd get that error. Something related to your hand-built GStreamer - but I think know why you'd end it would install a .la file that looks for a non-existent lib-libpthread

kevkid, 2019-06-19:
@thaytan Yeah, its weird. I built Gstreamer using cerbero and gstreamer 1.15

closed 2020-05-03 by thaytan

tp-m, 2020-05-03:
I think this got closed by mistake, I must've typoed an issue/mr number in my commit, sorry.

commit 588024c8 (2020-05-03): meson: drop map file and fix plugin symbol export with newer gstreamer versions

Use -fvisibility instead of a map file for symbol export, so that
the right symbols get exported with newer gstreamer versions. Older
GStreamer versions also still work of course.

Fixes blacklisting/plugin-loading issues with GStreamer >= 1.14

Fixes #84, closes #94 #67