thaytan/gst-rpicamsrc #85: webrtc streaming parameters

kevkid, 2019-02-11:
Hi, I am trying to use this project to stream my rpi-camera to my browser using webrtc. Looking at the webrtc-unidirectional example I can see your pipeline looks like this:

```
rpicamsrc bitrate=600000 annotation-mode=12 preview=false ! video/x-h264,profile=constrained-baseline,width=640,height=360,level=3.0 ! queue max-size-time=100000000 ! h264parse ! rtph264pay config-interval=-1 name=payloader ! application/x-rtp,media=video,encoding-name=H264,payload=96
```

I am curious as to how you figured out what parameters were needed to stream to the browser. Currently I am using this: https://github.com/centricular/gstwebrtc-demos for my webrtc backend, and this is my pipeline:

```
webrtcbin name=sendrecv 
 rpicamsrc bitrate=600000 annotation-mode=12 preview=false keyframe-interval=2 ! video/x-h264,profile=constrained-baseline,width=1280,height=720 ! rtph264pay ! application/x-rtp,media=video,encoding-name=H264,payload=96 ! sendrecv.
```

but I get no output to my screen just this:

[image]

and it never seems to load. Does my pipeline look like it should work in chrome/firefox?

thaytan, 2019-02-11:
It took some trial and error to get the settings right. Add the level=3.0 parameter to the caps, and the config-interval=-1 to the h264parse. The browsers are very fussy about the H264 profile and level they receive, and the 2nd parameter makes sure GStreamer repeats the H.264 SPS/PPS periodically in case they are lost on the way to the browser at the start of the transmission.

kevkid, 2019-02-12:
Hrmm interesting, I have been running this and it seems like it never gets the stream, I also get these errors from gstreamer:

```
(python3.7:1582): GStreamer-WARNING **: 00:00:53.193: ../gst/gstpad.c:4292:gst_pad_chain_data_unchecked:<dtlssrtpdec0:sink> Got data flow before stream-start event

(python3.7:1582): GStreamer-WARNING **: 00:00:53.193: ../gst/gstpad.c:4297:gst_pad_chain_data_unchecked:<dtlssrtpdec0:sink> Got data flow before segment event
```

I assume this is causing my stream to never play. Any idea what I can do about it?

Here is my pipeline:
```
PIPELINE_DESC = '''
webrtcbin name=sendrecv 
 rpicamsrc bitrate=600000 annotation-mode=12 preview=false ! video/x-h264,profile=constrained-baseline,width=640,height=360,level=3.0 ! queue max-size-time=100000000 ! h264parse ! rtph264pay config-interval=-1 ! application/x-rtp,media=video,encoding-name=H264,payload=96 ! sendrecv.
'''
```

kevkid, 2019-02-18:
@thaytan Hi, I was curious as to what browser you tried the webrtc demo on? I have gotten nowhere with this part. It never shows any video on my browser (ff and chrome). I tried your settings, and I have tried many different profiles to no avail. Sorry to keep bugging yah, but it seems like this is a problem that that has little documentation online.

thaytan, 2019-02-19:
One failure I've had that doesn't yet get caught is the dependency on the GStreamer nice elements (gstreamer1.0-nice package). If they're not present, you get a quite failure that only shows up in the debug logs.

I've used firefox, Chrome and Safari with it in the past (which is not to say that something didn't break). Most recently I tested with Firefox.