thaytan/gst-rpicamsrc #86: sync the options names between raspivid and rpicamsrc

Abu-Abdullah, 2019-05-21:
i would recommend to keep the same spelling between raspivid and rpicamsrc e.g.

drc: 
med (raspivid)
medium (rpicamsrc)

image-effect:
solarise (raspivid)
solarize (rpicamsrc)

posterise (raspivid)
posterize(rpicamsrc)

and the rest.

i have an app that display these options to user and set it in the command line, hence it is not optimized to set a condition for raspivid and rpicamsrc and keep track of changes.

Abu-Abdullah, 2019-05-21:
and of course the same goes for  the option name itself e.g. 

preview=false (rpicamsrc)
-n (raspivid)

keyframe-interval=? (rpicamsrc)
-g ? (raspivid)

tp-m, 2019-05-21:
Thanks for the suggestion and the issue.

Personally, I don't see much benefit in keeping option names between the `raspivid` command line utility and `rpicamsrc` in sync, even less so where the option is cryptic for `raspivid` and easy to understand on `rpicamsrc`. Changing existing options or their spelling is also not really an, err, option, because it might break existing applications.

So for me this is a WONTFIX I'm afraid :)

Abu-Abdullah, 2019-05-22:
Ok. noted. 

Usually anyone will depend on raspivid documentation, hence he will run through their style of naming so it will easy to understand rpicamsrc if the naming convention is similar (and easy to code as well). rpicamsrc is a second layer.

Break existing applications is another thing.

closed 2019-05-22 by Abu-Abdullah