thaytan/gst-rpicamsrc #89: libgstrpicamsrc.so is in the black list

manuel-leiva, 2019-05-25:
Hi,

I was using gstreamer-1.10.5 and gst-rpicamsrc commit 4fc608e and everything was working ok, but I have updated gstreamer to 1.16.0 and now gst-rpicamsrc is added into the black list.

```
pi@raspberrypi:~$ gst-inspect-1.0 -b
Blacklisted files:
  libgstrpicamsrc.so

Total count: 1 blacklisted file
```

I have updated gst-rpicamsrc commit 4ee114f in order to use a more recent version but I got the same error, For some reason, the latest GStreamer version fails loading gst-rpicamsrc plugin.

Is this a known problem?

Thanks.

thaytan, 2019-05-25:
No, that's not expected. You can try forcing a rescan (touch the libgstrpicamsrc.so file to give it a new timestamp) and run gst-inspect-1.0 again, in case the failure was transient, or run gst-inspect-1.0 /path/to/libgstrpicamsrc.so directly on the .so to see what happens.

manuel-leiva, 2019-05-25:
Thanks for your fast answer.

I have applied the suggested procedure but I got the same result.
```
pi@raspberrypi:~ $ gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
Could not load plugin file: File "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so" is not a GStreamer plugin
```
Checking gst-inspect code source, the previous message is printed from this line:
https://gitlab.freedesktop.org/gstreamer/gstreamer/blob/1.16.0/gst/gstplugin.c#L820

The error is generated is because `g_module_symbol` returns `FALSE` here:
https://gitlab.freedesktop.org/gstreamer/gstreamer/blob/1.16.0/gst/gstplugin.c#L805
where `symname` is `"gst_plugin_rpicamsrc_get_desc"`

If this command line is executed for libgstvideo4linux2.so
`gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so`

The value returned by `g_module_symbol` is `TRUE` and the `symname` is `"gst_plugin_video4linux2_get_desc"`

Do you know what could be the cause of the error in `g_module_symbol`?

thaytan, 2019-05-26:
I don't know what's gone on there. The symbol it's looking for is a 'new style' plugin descriptor. It used to be called 'gst_plugin_desc' - and the plugin loader should fall back to trying that symbol.

nm -D /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so | grep plugin

should show one or other of the symbols

manuel-leiva, 2019-05-27:
When `g_module_symbol` returns FALSE using the symbol name `gst_plugin_rpicamsrc_get_desc` the function tries again to get a symbol pointer from the module but using the symbol name `"gst_plugin_desc"` and fails again.

This is the `nm -D` output
```
pi@raspberrypi:~ $ nm -D /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so | grep plugin
         U gst_plugin_register_static
```
Is this the expected output?

thaytan, 2019-05-28:
No, that's not what I expected. There should be at least one of those 2 symbols available for GStreamer to load the plugin. It should look more like:

nm -D /usr/lib64/gstreamer-1.0/libgstcoreelements.so  | grep plugin
0000000000016840 T gst_plugin_coreelements_get_desc
0000000000016850 T gst_plugin_coreelements_register
                 U gst_plugin_register_static

Something hasn't built right, but I'm not sure what yet.

manuel-leiva, 2019-05-29:
Sorry, I have a mistake in the output reported previously for `nm -D`, the correct output is this
```
pi@raspberrypi:~ $ nm -D /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstcoreelements.so | grep plugin
0000d018 T gst_plugin_coreelements_get_desc
0000d024 T gst_plugin_coreelements_register
         U gst_plugin_register_static
```
Therefore I have the same symbols reported by you.

I continue searching what is the problem.

thaytan, 2019-05-29:
No - you did it right. I used the coreelements plugin in my example because I didn't have easy access to a raspberry pi at the time. The rpicamsrc plugin should have the equivalent symbols named with 'rpicamsrc' instead of coreelements, and I don't know why they haven't. I'll need to get an rpi set up with 1.16 and see

manuel-leiva, 2019-05-30:
Ok, understood

This is what I have seen:

When I run the command 
`gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so`

The value returned by g_module_symbol is TRUE and the symbol name is `"gst_plugin_video4linux2_get_desc"` and this is one of the symbols displayed by `nm -D`

```
pi@raspberrypi:~ $ nm -D /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so | grep plugin
         U gst_plugin_add_dependency
         U gst_plugin_register_static
0000746c T gst_plugin_video4linux2_get_desc
00007478 T gst_plugin_video4linux2_register
```

Now, when I have run the same command for `libgstrpicamsrc.so` (`gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so`) The value returned by g_module_symbol is FALSE and the symbol name is `gst_plugin_rpicamsrc_get_desc`.

This symbol is in the plugin but it is not a dinamic symbol
```
pi@raspberrypi:~ $ nm /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so | grep plugin
0001c3e4 d gst_plugin_desc
         U gst_plugin_register_static
00006188 t gst_plugin_rpicamsrc_get_desc
000061a0 t gst_plugin_rpicamsrc_register
0000613c t plugin_init
```
From my side gst-rpicamsrc, gstreamer, glib and its dependencies were cross-compiled and I am using gstreamer-1.16.0, glib-2.60.0 and gcc-linaro-6.1.1-2016.08-x86_64_arm-linux-gnueabihf
I think the cross-compilation is not the problem since other plugins are working correctly but if you have some clue or question, please let me know.
Thanks.

mleivaf, 2020-05-02:
Hi @thaytan 

I have tested gst-rpicamsrc again using GStreamer 1.16 but this time I have used [gst-build](https://github.com/GStreamer/gst-build) in order to compile GStreamer natively but I still getting the same problem:

```
pi@raspberrypi:~ $ gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
Could not load plugin file: File "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so" is not a GStreamer plugin
pi@raspberrypi:~ $ gst-inspect-1.0 -b
Blacklisted files:
  libgstrpicamsrc.so

Total count: 1 blacklisted file
```
Do you have some suggestion about how to debug this problem?

Thanks.

manuel-leiva, 2020-05-02:
Hi @thaytan

I have tested gst-rpicamsrc again using GStreamer 1.16 but this time I have used gst-build in order to compile GStreamer natively but I still getting the same problem:
```
pi@raspberrypi:~ $ gst-inspect-1.0 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so
Could not load plugin file: File "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so" is not a GStreamer plugin
pi@raspberrypi:~ $ gst-inspect-1.0 -b
Blacklisted files:
  libgstrpicamsrc.so

Total count: 1 blacklisted file
Do you have some suggestion about how to debug this problem?
```
Thanks.

thaytan, 2020-05-03:
Still a problem a year later! Unfortunately, I still don't know what is going wrong - something to do with the symbol exporting, but I'm not sure what.

manuel-leiva, 2020-05-03:
@thaytan,

I have seen the error is removed applying  the same change applied here: https://github.com/thaytan/gst-rpicamsrc/pull/94

Now the plugin is loaded by Gstreamer 1.16

Thanks.

thaytan, 2020-05-03:
Good news! Thanks - looks like we need to get the -fvisibility=hidden change done.

tp-m, 2020-05-03:
Should be fixed with #89.

manuel-leiva, 2020-05-07:
I am closing the issue since the error was fixed in https://github.com/thaytan/gst-rpicamsrc/commit/140664190716397f31cff1fb4cd6fa8eda17973f

Thanks

closed 2020-05-07 by manuel-leiva

thaytan, 2020-05-07:
Thanks for confirming!