thaytan/gst-rpicamsrc #92: mmal_component_create_core: could not find component 'vc.ril.camera - again

neilyoung, 2019-10-05:
RPI OS is Xenial 16.04, distribution
https://downloads.ubiquityrobotics.com/pi.html

Build and install fine:

gst-inspect-1.0 | grep rpicamsrc
rpicamsrc:  rpicamsrc: Raspberry Pi Camera Source
rpicamsrc:  rpicamsrcdeviceprovider (GstDeviceProviderFactory)

At runtime:

ubuntu@ubiquityrobot:~/gst-rtsp-server-1.8.3/examples$ ./test-launch --gst-debug=3 "( rpicamsrc bitrate=8000000 awb-mode=tungsten preview=false ! video/x-h264, width=640, height=480, framerate=30/1 ! h264parse ! rtph264pay name=pay0 pt=96 )"
stream ready at rtsp://127.0.0.1:8554/test
0:00:29.265774495 22236 0x7310de60 FIXME                default gstutils.c:3766:gst_pad_create_stream_id_internal:<appsrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
0:00:29.270842080 22236 0x75f098c0 ERROR              rpicamsrc RaspiCapture.c:1036:create_camera_component: Failed to create camera component
0:00:29.270904319 22236 0x75f098c0 ERROR              rpicamsrc RaspiCapture.c:1663:raspi_capture_setup: raspi_capture_setup: Failed to create camera component
0:00:29.271168328 22236 0x75f098c0 WARN                 basesrc gstbasesrc.c:3354:gst_base_src_start:<rpicamsrc0> error: Failed to start
0:00:29.271394578 22236 0x75f098c0 WARN                 basesrc gstbasesrc.c:3725:gst_base_src_activate_push:<rpicamsrc0> Failed to start in push mode
0:00:29.271458900 22236 0x75f098c0 WARN                GST_PADS gstpad.c:1106:gst_pad_set_active:<rpicamsrc0:src> Failed to activate pad
0:00:29.274027953 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2552:start_preroll: failed to preroll pipeline
0:00:29.274119827 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2675:start_prepare: failed to preroll pipeline
0:00:29.274646700 22236   0xec6690 WARN               rtspmedia rtsp-media.c:2573:wait_preroll: failed to preroll pipeline
0:00:29.274737533 22236   0xec6690 WARN               rtspmedia rtsp-media.c:2877:gst_rtsp_media_prepare: failed to preroll pipeline
0:00:29.275399823 22236 0x75f098c0 WARN               rtspmedia rtsp-media.c:2278:default_handle_message: 0x75f20198: got error GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (gstbasesrc.c(3354): gst_base_src_start (): /GstPipeline:media-pipeline/GstBin:bin0/GstRpiCamSrc:rpicamsrc0:
Failed to start)
0:00:29.286751966 22236   0xec6690 ERROR             rtspclient rtsp-client.c:763:find_media: client 0xdc2b48: can't prepare media
0:00:29.288495241 22236   0xec6690 ERROR             rtspclient rtsp-client.c:2288:handle_describe_request: client 0xdc2b48: no media
^C

Local test is also not working:

ubuntu@ubiquityrobot:~/gst-rpicamsrc$ gst-launch-1.0 rpicamsrc bitrate=1000000 ! filesink location=test.h264
Setting pipeline to PAUSED ...
mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure.
Additional debug info:
gstbasesrc.c(3354): gst_base_src_start (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
Failed to start
Setting pipeline to NULL ...
Freeing pipeline ...
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ 

This project seems to be not maintained anymore, but I wanted to try at least

thaytan, 2019-10-06:
Does raspivid work? If not, you haven't enabled the camera in the boot config.

neilyoung, 2019-10-06:
Yeah, would be nice if it would be that easy. Raspivd works like a champ

thaytan, 2019-10-07:
This seems similar to a problem that some other people have reported then, but I don't think anyone has yet come back with a solution, and I've never come across it.

I build and test on raspbian, so it may be a difference in how the Xenial distro ships the raspberry pi support libs

neilyoung, 2019-10-07:
@thaytan Right, I think you are referring to this report https://github.com/thaytan/gst-rpicamsrc/issues/51

Indeed, that's the case. Any chance to approach the problem once more?

thaytan, 2019-10-07:
Sure.
How did you build rpicamsrc? Maybe there's something different there.

neilyoung, 2019-10-07:
Thanks. Very much appreciated.

Basically this is the build chain:

```
git clone https://github.com/thaytan/gst-rpicamsrc.git
cd gst-rpicamsrc/
./autogen.sh
make
sudo make install
```

neilyoung, 2019-10-08:
@thaytan Any comments on this?

thaytan, 2019-10-09:
That's how I do it. I suspect Ubuntu must package / link the libs differently somehow. Can you attach the config.log file from the build dir?

neilyoung, 2019-10-09:
Here you are:

[config.log](https://github.com/thaytan/gst-rpicamsrc/files/3707361/config.log)

neilyoung, 2019-10-11:
Hmm, how did you get that information? `make` or `sudo make install` doesn't give that.
I can't even find `libbcm_host.so` anywhere in the outputs.

neilyoung, 2019-10-11:
```
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ ldd -r /opt/vc/lib/libbcm_host.so
	linux-vdso.so.1 =>  (0x7efcc000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f5f000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76f0d000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76ef4000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ed0000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76ebd000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76ea7000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76dbb000)
	/lib/ld-linux-armhf.so.3 (0x76f9b000)

```

neilyoung, 2019-10-11:
Seems, this is missing in my log:

`/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so (0x76ef4000)`

thaytan, 2019-10-12:
I don't know what to make of that either :-/

Here's a couple more things we could compare - ldd output of libgstrpicamsrc.so vs raspivid on raspbian:

ldd -r gst-rpicamsrc/src/.libs/libgstrpicamsrc.so 
	linux-vdso.so.1 (0x7ee59000)
	/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so (0x76f54000)
	libgstvideo-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstvideo-1.0.so.0 (0x76e68000)
	libgstbase-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstbase-1.0.so.0 (0x76dfa000)
	libgstreamer-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstreamer-1.0.so.0 (0x76cd7000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0x76c7d000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76b75000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76b57000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76b37000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0x76b1c000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76b03000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76adc000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ab3000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76974000)
	/lib/ld-linux-armhf.so.3 (0x76f31000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x768fc000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x7687d000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgmodule-2.0.so.0 (0x76869000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76852000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x7683f000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x76827000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x767ae000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76796000)
	libvcsm.so => /opt/vc/lib/libvcsm.so (0x76781000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76754000)

ldd -r /usr/bin/raspivid 
	linux-vdso.so.1 (0x7ed40000)
	/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so (0x76f4e000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76eea000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76eca000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0x76eaf000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76e88000)
	libvcsm.so => /opt/vc/lib/libvcsm.so (0x76e73000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76e5d000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76e44000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e1b000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76e08000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76df1000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76cb2000)
	/lib/ld-linux-armhf.so.3 (0x76f2b000)

neilyoung, 2019-10-12:
libmmal_vc_client is missing in the first dump

```
ubuntu@ubiquityrobot:~$ ldd -r gst-rpicamsrc/src/.libs/libgstrpicamsrc.so
	linux-vdso.so.1 =>  (0x7ef9a000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76e8a000)
	libgstvideo-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstvideo-1.0.so.0 (0x76de8000)
	libgstbase-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstbase-1.0.so.0 (0x76d91000)
	libgstreamer-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstreamer-1.0.so.0 (0x76cb6000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0x76c6e000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76b96000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76b78000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76b58000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76b3f000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76b18000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76af4000)
	/lib/ld-linux-armhf.so.3 (0x76f95000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x76a95000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76a1d000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgmodule-2.0.so.0 (0x76a0a000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x769f4000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x769e1000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x769cb000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x7696e000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76958000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76930000)

```

```
ubuntu@ubiquityrobot:~/gst-rpicamsrc$  ldd -r /usr/bin/raspivid
	linux-vdso.so.1 =>  (0x7ef31000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76e89000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76e2f000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76e0f000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0x76df4000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76dcd000)
	libvcsm.so => /opt/vc/lib/libvcsm.so (0x76db8000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76da2000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76d89000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76d65000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76d52000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76d3c000)
	/lib/ld-linux-armhf.so.3 (0x76f75000)

```

thaytan, 2019-10-12:
Aha! I think the missing libmmal_vc_client.so is key - it's supposed to be there. Is it in the LIBADD line in src/Makefile.am?

> libgstrpicamsrc_la_LIBADD = $(GST_LIBS) $(RPI_LIBFLAGS) -lmmal_core -lmmal_util -lmmal_vc_client -lvcos -lbcm_host

neilyoung, 2019-10-12:
Seems so:

`libgstrpicamsrc_la_LIBADD = $(GST_LIBS) $(RPI_LIBFLAGS) -lmmal_core -lmmal_util -lmmal_vc_client -lvcos -lbcm_host
`

thaytan, 2019-10-12:
I think I see - seems Ubuntu might add --as-needed to the linker flags by default. https://wiki.ubuntu.com/NattyNarwhal/ToolchainTransition. Let me make a patch for you to test.

neilyoung, 2019-10-12:
OK. Will be out for an hour or so. Take your time

thaytan, 2019-10-12:
It turns out this is a whole rabbit hole. Adding -Wl,--no-as-needed doesn't work because libtool reorders the linker arguments and puts it in the wrong place. There's been a bug open about that since 2006 - https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=347650

I'm not sure how to get that done, or why things work when they build the raspivid binary.

https://launchpad.net/~ubuntu-pi-flavour-makers/+archive/ubuntu/ppa/+sourcepub/7476907/+listing-archive-extra is the raspberrypi-firmware deb source (sticking a link here so I can find it again later)

neilyoung, 2019-10-12:
Stupid question: What if your code references a harmless function from mmal_vc_client? Wouldn't that force the linkage?

thaytan, 2019-10-12:
one thing that would be easy to try: install meson and build using that.
It doesn't use libtool and (at least here) doesn't add --as-needed to the link flags.
>>>
sudo apt-get install meson
meson build
ninja -C build -v
>>>

neilyoung, 2019-10-12:
Looks good, doesn't work :)

First problem: `build` dir required (?)

```
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ meson build
Error during basic setup:

[Errno 2] No such file or directory: '/home/ubuntu/gst-rpicamsrc/build'

```
Solved:

`ubuntu@ubiquityrobot:~/gst-rpicamsrc$ mkdir build
`

Then:

```
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ meson build
The Meson build system
Version: 0.29.0
Source dir: /home/ubuntu/gst-rpicamsrc
Build dir: /home/ubuntu/gst-rpicamsrc/build
Build type: native build
Build machine cpu family: arm
Build machine cpu: armv7l

Meson encountered an error in file meson.build, line 1, column 0:
Meson version is 0.29.0 but project requires >= 0.34.0.
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ 

```

neilyoung, 2019-10-12:
OK, could work around by installing meson via pip3:

```
pip3 install --user meson
 ~/.local/bin/meson build

```

ninja did run, and there is a `libgstrpicamsrc.o` in `./build/src`, but the result seems missing the lib too:
```

ldd -r libgstrpicamsrc.so 
	linux-vdso.so.1 =>  (0x7ede6000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76ea3000)
	libgstreamer-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstreamer-1.0.so.0 (0x76d8c000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0x76d44000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76c6c000)
	libgstbase-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstbase-1.0.so.0 (0x76c15000)
	libgstvideo-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstvideo-1.0.so.0 (0x76baf000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76b91000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76b71000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76b58000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76b31000)
	/lib/ld-linux-armhf.so.3 (0x76fb0000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgmodule-2.0.so.0 (0x76b1e000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76aa6000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76a90000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76a7d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76a59000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x76a43000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x769e6000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x76987000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76971000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76949000)

```

thaytan, 2019-10-12:
Oh dear. Seems like your ubuntu has an even older meson than I do (My Raspbian has 0.37.1) and then the newer one pip3 installs might be adding --as-needed by default too.
Can you put the ninja -v output (you'll need to 'ninja clean' first)

neilyoung, 2019-10-12:
:) No, I guess, 0.29 is default for 16.04 LTS. There is a backport of 0.40, but I didn't want to start messing with this. 

Here is a related bug https://github.com/Motion-Project/motion/issues/351

I couldn't see that linker flag at the first glance, but I might have overseen it.

```
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ ninja -C build -v
ninja: Entering directory `build'
[1/10] /home/ubuntu/.local/bin/meson --internal exe --capture src/gstrpicam-enum-types.h -- /usr/bin/glib-mkenums --template /home/ubuntu/gst-rpicamsrc/src/gstrpicam-enums-template.h ../src/gstrpicam_types.h
[2/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrcdeviceprovider.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrcdeviceprovider.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrcdeviceprovider.c.o' -c ../src/gstrpicamsrcdeviceprovider.c
[3/10] /home/ubuntu/.local/bin/meson --internal exe --capture src/gstrpicam-enum-types.c -- /usr/bin/glib-mkenums --template /home/ubuntu/gst-rpicamsrc/src/gstrpicam-enums-template.c ../src/gstrpicam_types.h
[4/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrc.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrc.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrc.c.o' -c ../src/gstrpicamsrc.c
[5/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/RaspiCapture.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/RaspiCapture.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/RaspiCapture.c.o' -c ../src/RaspiCapture.c
[6/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/RaspiCamControl.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/RaspiCamControl.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/RaspiCamControl.c.o' -c ../src/RaspiCamControl.c
[7/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/RaspiPreview.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/RaspiPreview.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/RaspiPreview.c.o' -c ../src/RaspiPreview.c
[8/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/RaspiCLI.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/RaspiCLI.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/RaspiCLI.c.o' -c ../src/RaspiCLI.c
[9/10] cc -Isrc/25a6634@@gstrpicamsrc@sha -Isrc -I../src -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreas -I/opt/vc/include/interface/vmcs_host/linux -I. -I../ -I/usr/include/gstreamer-1.0 -I/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -pthread -DHAVE_CONFIG_H -MD -MQ 'src/25a6634@@gstrpicamsrc@sha/meson-generated_.._gstrpicam-enum-types.c.o' -MF 'src/25a6634@@gstrpicamsrc@sha/meson-generated_.._gstrpicam-enum-types.c.o.d' -o 'src/25a6634@@gstrpicamsrc@sha/meson-generated_.._gstrpicam-enum-types.c.o' -c src/gstrpicam-enum-types.c
[10/10] cc  -o src/libgstrpicamsrc.so 'src/25a6634@@gstrpicamsrc@sha/meson-generated_.._gstrpicam-enum-types.c.o' 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrc.c.o' 'src/25a6634@@gstrpicamsrc@sha/gstrpicamsrcde [... cut ...] c/gstplugin.map /usr/lib/arm-linux-gnueabihf/libgstreamer-1.0.so /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so /usr/lib/arm-linux-gnueabihf/libglib-2.0.so /usr/lib/arm-linux-gnueabihf/libgstbase-1.0.so /usr/lib/arm-linux-gnueabihf/libgstvideo-1.0.so /opt/vc/lib/libmmal_core.so /opt/vc/lib/libmmal_util.so /opt/vc/lib/libmmal_vc_client.so /opt/vc/lib/libvcos.so /opt/vc/lib/libbcm_host.so -Wl,--end-group -Wl,-rpath,/opt/vc/lib -Wl,-rpath-link,/opt/vc/lib
ubuntu@ubiquityrobot:~/gst-rpicamsrc$ 

```

thaytan, 2019-10-12:
Thanks for finding the motion bug. The same solution doesn't work here because of an annoying wrinkle - libtool will reorder -Wl,--no-as-needed and ruin everything when linking a .so, but leaves them alone when linking a final binary like motion.

I also just discovered that I explicitly fixed this in the meson build in 2016 (commit #f5416e) - which is why the flag isn't in the ninja output. However the lib still gets dumped from the output, so I suspect Ubuntu really have made --as-needed the default in ld

I've just pushed a patch for the meson build to the no-as-needed branch to explicitly add --no-as-needed. Can you please test that?

neilyoung, 2019-10-12:
...and tadaaa: Here we are :)

```
ubuntu@ubiquityrobot:~/gst-rpicamsrc/build/src$ ldd -r libgstrpicamsrc.so 
	linux-vdso.so.1 =>  (0x7ef9e000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76dcd000)
	libgstreamer-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstreamer-1.0.so.0 (0x76cb6000)
	libgobject-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0x76c6e000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76b96000)
	libgstbase-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstbase-1.0.so.0 (0x76b3f000)
	libgstvideo-1.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgstvideo-1.0.so.0 (0x76ad9000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0x76abb000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0x76a9b000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0x76a80000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76a67000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76a40000)
	/lib/ld-linux-armhf.so.3 (0x76eda000)
	libgmodule-2.0.so.0 => /usr/lib/arm-linux-gnueabihf/libgmodule-2.0.so.0 (0x76a2d000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x769b5000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x7699f000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x7698c000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76968000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x76952000)
	libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x768f5000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x76896000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76880000)
	libvcsm.so => /opt/vc/lib/libvcsm.so (0x7686b000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76843000)
ubuntu@ubiquityrobot:~/gst-rpicamsrc/build/src$ 

```

Where to copy the resulting .so?

thaytan, 2019-10-12:
If you build with 'meson --prefix=/usr' then ninja install should do the right thing, but you can just copy to  /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/ is the easiest way

You can also do 'GST_PLUGIN_DIR=`pwd` gst-launch-1.0 rpicamsrc ! ...' to use the plugin file in place.

neilyoung, 2019-10-12:
Yepp. Did it all from scratch again. Works. I had to run `sudo ninja install` since meson was installed with user privileges, but it worked.

The test command produced a test.h264 file. Checking validity right now, but I'm confident.

Good job, thanks. Will your merge that change to master?

thaytan, 2019-10-12:
I will, and I'll think about what to do with the autotools build here.

neilyoung, 2019-10-12:
Cool. Thanks for your efforts. Have a nice weekend!

thaytan, 2019-10-12:
Thanks for running that down!

Meson build fix merged in 79860a0b2a0f3beef6ccfab0cf7d531e55e3b06c

thaytan, 2019-10-12:
I could just delete the autotools build...