thaytan/gst-rpicamsrc #93: Internal error when trying to build it

jeremyfritzen, 2019-11-03:
Hi,

I tried to install this code on my **Raspberry Pi B2** (**_Raspbian GNU/Linux 10 (buster)_**)
I followed the instructions on README (including `apt-get install autoconf automake libtool pkg-config libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libraspberrypi-dev`, which ends successfully) but **I get the following error when trying `./autogen.sh --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf/`:**

> autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
/usr/bin/m4: internal error detected; please report this bug to <bug-m4@gnu.org>: Illegal instruction
autom4te: /usr/bin/m4 failed with exit status: 2
aclocal: error: echo failed with exit status: 2
autoreconf: aclocal failed with exit status: 2
autogen.sh failed

I also tried to run the command with sudo but it doesn't work.
Do you have any idea on how to fix it?

Thanks a lot!

thaytan, 2019-11-03:
That... looks like your system is not installed right, or it's overheating or something.

jeremyfritzen, 2019-11-03:
Thank you for your answer!
How can you see that the raspberry is overheating?

thaytan, 2019-11-03:
/usr/bin/m4: internal error detected; please report this bug to bug-m4@gnu.org: Illegal instruction

That just shouldn't happen - either your m4 binary is corrupted, or it was installed in a weird way and is the wrong architecture, or something even more sinister is happening

jeremyfritzen, 2019-11-04:
Thank you.
Never heard about m4. Whats is it for?
It's weird because I just installed the last version on raspbian on it...

thaytan, 2019-11-04:
m4 is a processing tool that many open source projects depend on. It definitely shouldn't be crashing. I don't think this is a rpicamsrc specific problem - but something wrong with your SD card, or your board. Are you seeing any other stability issues?

jeremyfritzen, 2019-11-06:
I didn't see any other issues so far. I just installed the last version of raspbian for this software.
I will try to re-install it.

jeremyfritzen, 2019-11-13:
Ok it works now! Thanks for your help :)
I still have an error but it's a different topic. I can close this one.

closed 2019-11-13 by jeremyfritzen