thaytan/gst-rpicamsrc #95: redirect preview

joukeHijlkema, 2020-03-28:
Hi,
To monitor experiments in our lab I've several cameras streaming live video to a server. When given the order these streams are recoded on the server. This strains the network to much when the number of cameras grows and it would be preverable to stream low-res preview footage to the server and record hig-res videos on the rpi it self. When finished the server can get the videos one by one. 
My question is thus; can I (mis) use the rpicamsrc preview as the low-res source end stream it directly to a server ?
Thanks for your work and your help,
Jouke

thaytan, 2020-03-29:
There's a couple of considerations:
1) At the moment, no - the internal preview port outputs directly to a hardware overlay onto the screen
2) I've never experiemented with doing anything else with it, and ability to do that depends a lot on the RPi videocore firmware
3) To get a low-resolution preview, you'd need to rescale and encode, so that's a chunk of extra videocore resources required
4) There's some GStreamer considerations around getting 2 streams out of rpicamsrc, but that's just extra work.

What one could definitely do would be to encode with say 1 keyframe per second, use tee to duplicate the H.264 stream for recording and for streaming, and then use a pad probe on the streaming branch to drop everything except the keyframes. You'd get a lower-bandwidth 1fps stream consisting only of keyframes.

joukeHijlkema, 2020-03-29:
That's an exelent idea ! Thank you jan.