thaytan/gst-rpicamsrc #102: rpicamsrc not working with gst_parse_launch

Rpi112, 2020-07-16:
Hi,

I am using the following C code in a program to start the pipeline.

pipeline = gst_parse_launch( rpicamsrc ! queue ! videoconvert ! video/x-raw,width=1280,height=720,framerate=[30/1,10000000/333333] ! x264enc bframes=0 speed-preset=veryfast bitrate=512 byte-stream=TRUE tune=zerolatency ! video/x-h264,stream-format=byte-stream,alignment=au,profile=baseline ! appsink sync=TRUE emit-signals=TRUE                     name=appsink-video, &error);

This is not giving any readable output. Are there any errors in above code?

thaytan, 2020-07-16:
the point of rpicamsrc is that it directly outputs H.264 video using the hardware video encoder. You can do `rpicamsrc bitrate=512000 ! video/x-h264,width=1280,height=720,framerate=30/1,profile=baseline ! appsink sync=TRUE emit-signals=TRUE name=appsink-video`

Rpi112, 2020-07-16:
Oh, got it. Thanks.. It's working now...

closed 2020-07-16 by Rpi112