thaytan/gst-rpicamsrc #105: "Camera capture timed out." when setting pipeline from GST_STATE_NULL to GST_STATE_PLAYING for the second time

TimSchneider42, 2020-11-12:
Hey,

I have a little project where I use a Raspberry Pi as a surveillance camera. In order to save some energy (as the Pi is powered from a powerbank) I use a motion sensor to disable the IR LED and stop querying the camera whenever there is no motion in the area.

Since constructing my pipeline takes a lot of time (~1s), I only construct it once in the beginning and set it to GST_STATE_NULL while I am not using it. On my desktop PC, I use the following pipeline:

``` v4l2src device=/dev/video0 ! videoconvert ! x264enc ! h264parse ! splitmuxsink [...] ```

and everything works fine.

On my Pi on the other hand I use 

``` rpicamsrc bitrate=1000000 ! h264parse ! splitmuxsink [...] ```

which works fine the first time I start the pipeline, but yields the following error on the second time:

```
Error received from element raspicam: Camera capture timed out.
Debugging information: gstrpicamsrc.c(1445): gst_rpi_cam_src_create (): /GstPipeline:raspiwatch-pipeline/GstRpiCamSrc:raspicam:
Waiting for a buffer from the camera took too long.
Error received from element raspicam: Camera capture timed out.
Debugging information: gstrpicamsrc.c(1445): gst_rpi_cam_src_create (): /GstPipeline:raspiwatch-pipeline/GstRpiCamSrc:raspicam:
Waiting for a buffer from the camera took too long.
Error received from element raspicam: Internal data stream error.
**
ERROR:gstsplitmuxsink.c:1380:handle_gathered_gop: assertion failed: (queued_gop_time >= 0)
```

An example that reproduces the problem can be found here: <https://pastebin.com/ZvYpbeyj> and the full debug output of that example here: <https://pastebin.com/XET7ZyD2>.

Do you have any idea why that happens?

Best,
Tim

thaytan, 2020-11-12:
The splitmuxsink problem is (hopefully) fixed in git master, but is triggered in this case by the lack of video input when the camera times out. That's not a condition that I've tested - it could be that there's something not being cleaned up right (although I've been quite careful about that), but it could also be that there's something in the underlying MMAL implementation and driver that doesn't get cleaned up completely and prevent reuse.

I can think of 2 useful pieces of diagnostic info:
1) GST_DEBUG=rpi*:7 output - and compare what happens on the 2nd run vs the first.
2) Delete the entire pipeline and recreate it for the 2nd run - if that works, it's more likely a problem in rpicamsrc than in the underlying MMAL.

TimSchneider42, 2020-11-12:
Hey, 

thanks for your quick reply. I did both of the things you suggested.
1. Setting the log level to rpi*:7 yields the following output: <https://pastebin.com/s6aM6Q5v>. I made it such that before each run "=======...===" is printed.
2. Deleting the entire pipeline after each run works, the error does not occur.

Do you need any more information to debug this?

thaytan, 2020-11-13:
The pastebin log disappeared already unfortunately. Can you attach it here?

TimSchneider42, 2020-11-13:
Oh, sorry I thought it stayed up there for a while:

```
0:00:00.005846000  1040   0x9a0a80 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.019447000  1040   0x9a0a80 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.025571000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.030487000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.033396000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.036144000  1040   0x9a0a80 INFO            GST_REGISTRY gstregistry.c:1727:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.352634000  1040   0x9a0a80 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.arm.bin in 0.313041 seconds
0:00:00.355654000  1040   0x9a0a80 INFO            GST_REGISTRY gstregistry.c:1583:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.380423000  1040   0x9a0a80 INFO            GST_REGISTRY gstregistry.c:1685:scan_and_update_registry: Registry cache has not changed
0:00:00.383173000  1040   0x9a0a80 INFO            GST_REGISTRY gstregistry.c:1762:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.385891000  1040   0x9a0a80 INFO                GST_INIT gst.c:807:init_post: GLib runtime version: 2.58.3
0:00:00.388678000  1040   0x9a0a80 INFO                GST_INIT gst.c:809:init_post: GLib headers version: 2.58.1
0:00:00.390239000  1040   0x9a0a80 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.425792000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrpicamsrc.so" loaded
0:00:00.428765000  1040   0x9a0a80 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "rpicamsrc" named "raspicam"
0:00:00.442872000  1040   0x9a0a80 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0xa92ed0> adding pad 'src'
0:00:00.467362000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:00.470422000  1040   0x9a0a80 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "h264parse" named "h264parse"
0:00:00.474781000  1040   0x9a0a80 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x9fade8> adding pad 'sink'
0:00:00.477860000  1040   0x9a0a80 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x9fade8> adding pad 'src'
0:00:00.481347000  1040   0x9a0a80 INFO               baseparse gstbaseparse.c:3961:gst_base_parse_set_pts_interpolation:<GstH264Parse@0x9fade8> PTS interpolation: no
0:00:00.521706000  1040   0x9a0a80 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstmultifile.so" loaded
0:00:00.524561000  1040   0x9a0a80 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "splitmuxsink" named "splitmuxsink"
0:00:00.529480000  1040   0x9a0a80 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "pipeline" named "raspiwatch-pipeline"
0:00:00.533718000  1040   0x9a0a80 INFO        GST_ELEMENT_PADS gstutils.c:1774:gst_element_link_pads_full: trying to link element raspicam:(any) to element h264parse:(any)
0:00:00.536610000  1040   0x9a0a80 INFO                GST_PADS gstutils.c:1035:gst_pad_check_link: trying to link raspicam:src and h264parse:sink
0:00:00.539573000  1040   0x9a0a80 INFO                GST_PADS gstpad.c:4232:gst_pad_peer_query:<h264parse:src> pad has no peer
0:00:00.542465000  1040   0x9a0a80 INFO                GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: raspicam and h264parse in same bin, no need for ghost pads
0:00:00.544377000  1040   0x9a0a80 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link raspicam:src and h264parse:sink
0:00:00.546243000  1040   0x9a0a80 INFO                GST_PADS gstpad.c:4232:gst_pad_peer_query:<h264parse:src> pad has no peer
0:00:00.546841000  1040   0x9a0a80 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked raspicam:src and h264parse:sink, successful
0:00:00.548602000  1040   0x9a0a80 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.549010000  1040   0x9a0a80 INFO               GST_EVENT gstpad.c:5808:gst_pad_send_event_unchecked:<raspicam:src> Received event on flushing pad. Discarding
0:00:00.549616000  1040   0x9a0a80 INFO        GST_ELEMENT_PADS gstutils.c:1774:gst_element_link_pads_full: trying to link element h264parse:(any) to element splitmuxsink:(any)
[... cut ...]
Error received from element raspicam: Internal data stream error.
Debugging information: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:raspiwatch-pipeline/GstRpiCamSrc:raspicam:
streaming stopped, reason error (-5)
Error received from element raspicam: Internal data stream error.
Debugging information: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:raspiwatch-pipeline/GstRpiCamSrc:raspicam:
streaming stopped, reason error (-5)
**
ERROR:gstsplitmuxsink.c:1380:handle_gathered_gop: assertion failed: (queued_gop_time >= 0)

```

thaytan, 2020-11-13:
Hmm, looks like I forgot to set `src->started = FALSE;` in `gstrpicamsrc.c:gst_rpi_cam_src_stop` after cleanup - so it's not setting up the MMAL pieces on the 2nd run. Can you give that a try?

TimSchneider42, 2020-11-18:
That worked, I created a pull request

commit 44eec2f2 (2020-11-19): Added "src->started = FALSE;" to gst_rpi_cam_src_stop

Makes the element reusable multiple times after a state change back to READY.
Fixes #105
closed 2020-11-19 by thaytan