thaytan/gst-rpicamsrc #108: Typo in log message

PanderMusubi, 2021-01-25:
Sorry to post this here. No way of reaching the user @Consti10 directly that made a fork that probably will make a merge request (looking forward to that).

There is a typo in the error log message https://github.com/Consti10/gst-rpicamsrc/commit/4eb76a67fee640c0874d8210682e4224a3d784e1#diff-3feb97e9fa6f4d8dedd57d1db1f6ce44c27a4c01ac5b4e2ba3ce1368645a29beR1436

Consti10, 2021-01-25:
Thanks.
I can make a mrege request, however, note that according to https://github.com/OpenHD/Open.HD/issues/406
Sliced encoding rather increases latency than decreasing it, making it useless for our use case.

PanderMusubi, 2021-01-25:
Well, only if you think it is a useful addition. I'm not the maintainer here, just a power user that was looking into all the available branches for improvements. Then I saw this https://github.com/thaytan/gst-rpicamsrc/issues/100 . So a PR might be a MR there.

Consti10, 2021-01-25:
Yes, I am still waiting for an answer from the rpi foundation, maybe I am missing a parameter. I don't see any other use case for sliced encoding than to reduce latency.