thaytan/gst-rpicamsrc #109: is rpicamsrc really available with gstreamer v1.18  ?

lakis1982, 2021-02-22:
Hello @thaytan ,

I have installed on my raspberry pi 3 the gstreamer v1.18.3. But when i hit  
**gst-inspect-1.0 rpicamsrc**     i get the error
**No such element or plugin 'rpicamsrc'**

As far as I understood the _rpicamsrc_ should be available with gstreamer v1.18. What should I do to solve this problem ? Why does this happen ? 

Thank you

thaytan, 2021-02-23:
The rpicamsrc code is merged into `gst-plugins-good`, but that doesn't mean that it's been packaged in Raspbian yet. Or are you building GStreamer yourself?

lakis1982, 2021-02-23:
@thaytan thanks for your reply

I must mention that i dont have any knowledge of programming, compiling, building sources etc.. THese are like chinese to me.. 
To your question, i must say that I have raspbian Buster... It already had gstreamer 1.14 when installed the buster image.. What I did was to go in the      **/etc/apt/sources.list**      and  in the line (this is only one repository inside the sources.list)        _deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi_       change the word  buster       **buster**    with       **bullseye** .     Then i did  a     _sudo apt-get update_       and         _sudo apt-get dist-upgrade_        and Gstreamer was updated to v1.18..... 
What should I do now in order to have rpicamsrc   as well ? 

Thanks a lot

tweirtx, 2021-08-23:
Now that Bullseye is stable, I have installed it on my Pi and can confirm that this does not appear to be working. Any time I try to create a pipeline with rpicamsrc it fails with a `no element "rpicamsrc"` error. I have confirmed that the version of gstreamer I have installed is 1.18.4

thaytan, 2021-08-23:
rpicamsrc is definitely in the gst-plugins-good repository. If it's not being found, that means the plugin wasn't built - or if you''re installing from packages, that it isn't in the packages you've installed, or maybe isn't being packaged in the distro.

If you're building from source, confirm in the meson output from gst-plugins-good that rpicamsrc is enabled.
If you're installing from packages, then you either need to install something extra, or else complain to Raspbian that they haven't packaged the element.

I'm not planning on updating this external repo.

closed 2021-08-23 by thaytan

CNugteren, 2022-06-15:
Not really a solution to this issue, but for others that end up here searching for `rpicamsrc` support or their PiCam: on the latest 64-bit Raspberry Pi OS (Bullseye 11 at the moment): you can alternatively use [the `libcamerasrc` filter](https://libcamera.org/getting-started.html#using-gstreamer-plugin).