thaytan/gst-rpicamsrc #112: mmal: mmal_component_create_core: could not find component 'vc.ril.camera' Ubuntu mate 20.04 64 bit RPi 4B

fjp, 2021-06-03:
Hi, on a Raspberry Pi 4B running Ubuntu Mate 20.04 64 bit I am getting the following error when trying to run `gst-launch-1.0 rpicamsrc bitrate=1000000 ! filesink location=test.h264` 

```
Setting pipeline to PAUSED ...
mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure.
Additional debug info:
gstbasesrc.c(3468): gst_base_src_start (): /GstPipeline:pipeline0/GstRpiCamSrc:rpicamsrc0:
Failed to start
Setting pipeline to NULL ...
Freeing pipeline ...
```

`raspivid` is working and with the following I can stream a video from the RPi:

``` 
raspivid -t 0 -h 720 -w 1080 -fps 25 -hf -b 2000000 -o - | gst-launch-1.0 -v fdsrc ! h264parse !  rtph264pay config-interval=1 pt=96 ! gdppay ! tcpserversink host=YOUR_RPI_IP_ADDRESS port=5000
```

I have installed rpicamsrc following the instructions in the readme:

```
./autogen.sh --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf/
make
sudo make install
```

And I also installed with meson/ninja explained [here](https://github.com/thaytan/gst-rpicamsrc/issues/92#issuecomment-541322167) in #92. 

My GStreamer version is 1.16.2

Do you have any hints on how to get this working?

tprins228, 2021-06-07:
I've got the same issue while trying to install and run this in a 20.04 ubuntu arm7 docker image.

tp-m, 2021-06-07:
Possible that some of the recent commits in https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/-/commits/master/sys/rpicamsrc might be relevant, up to including https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/-/commit/477eba2dd7aff757f34566af28bf2e77c5e4227b (there were some follow-up build fixes to that).